ipmitool/ipmitool #6: Set password is not working with user id 16

sll552, 2018-05-15:
Hello,
I am using CentOS 7.4 recently ipmitool got upgraded to 1.8.18 from 1.8.15. This broke setting the password for a User with the id 16 with the following error:
`IPMI command failed: Invalid data field in request`

The corresponding code is https://github.com/ipmitool/ipmitool/blob/master/lib/ipmi_user.c#L208 .
```
memset(data, 0, data_len);
data[0] = (is_twenty_byte) ? 0x80 : 0x00;
data[0] |= (0x0F & user_id);
data[1] = 0x03 & operation;
```
`0x0F & 16` will result in 0 and is therefore not working. As I don't know the ipmi protocol well enough to change it myself I hope this gives the right clues for fixing this (and maybe get it backported in rhel/centos).

Best regards.

AlexanderAmelkin, 2018-05-16:
This looks like a bug. According to IPMI 2.0 specification, the User ID field is 5-bit, so the mask should be 0x3F, not 0x0F. I'll fix it.

AlexanderAmelkin, 2018-05-16:
@sll552, please check if branch bugfix/6 works for you. I don't have hardware to check against.

sll552, 2018-05-18:
Thanks for the fast fix!
It is working on my test system (Dell R620):
```
./ipmitool user set password 16
Password for user 16:
Password for user 16:
Set User Password command successful (user 16)
```
BTW: 
I had problems building the source in an environment without direct internet access. It seems that the generated pen file is not properly clean when using `make clean` and an invalid file is generated when no  direct internet connection is available, so curl fails. 
Also a little bit of "How to build" documentation would be nice.

closed 2018-05-30 by AlexanderAmelkin