ipmitool/ipmitool #7: Method to obtain the system uuid from the system?

rustyscottweber, 2018-05-16:
Oh gracious masters of the all-powerful ipmitool!  Is there a way to read the system’s uuid using ipmi so that we can perform PXE boots without logging into the system?  I have several threads on several different places and I would only contact you as a last resort…  The information is in the DMI table, I just am not sure how to use the tools I have to read it and reading the mac addresses and the uuid would be powerful allies in our quest to automate everything.

chiabinglee, 2018-05-16:
Maybe you could use IPMI "Get System GUID" command below:
```
ipmitool raw 0x06 0x37
ipmitool mc guid
```

Get MAC Address command:
```
ipmitool raw 0x0c 0x02 0x01 0x05 0x00 0x00
```
The details could be refer to the IPMI Spec "22.14 Get System GUID Command"
https://www.intel.com/content/dam/www/public/us/en/documents/product-briefs/ipmi-second-gen-interface-spec-v2-rev1-1.pdf
IPMI commands usage could refer to:
https://computercheese.blogspot.tw/2013/05/ipmi-messaging-support-commands.html

AlexanderAmelkin, 2018-05-16:
Please note that "Get System GUID" command is optional. Not all BMCs may have it implemented.

closed 2018-05-16 by AlexanderAmelkin

rustyscottweber, 2018-06-05:
It took me some time to process this and the question remains unanswered.
According to:
https://www.intel.com/content/dam/www/public/us/en/documents/product-briefs/ipmi-second-gen-interface-spec-v2-rev1-1.pdf
In section 20.8
The BMC has a controller which has a GUID, and there is a command to obtain that data using the following commands:
```
ipmitool raw 0x06 0x37
ipmitool mc guid

```
However! The bios has a dmi table which includes things like FRU data as well as the serial number and also, most importantly, a product UUID which is a globally unique ID which identifies that system not the BMC.  The UUID that I am after is the smbios/dmi table UUID which is what ipxe will return.

This is discussed in section 22.14 of the above document, but there doesn't seem to be anywhere in the documentation on how execute the get_system_UUID, only the before mentioned management controller guid.

To reiterate the question: what is the command, if it exists, to obtain the system's smbios uuid?

chiabinglee, 2018-06-06:
DMI table could not be read by BMC directly.  
Maybe there are two methods to implement: 
One is to create an OEM command in both BIOS and BMC firmware.
Another is BIOS stores DMI information as FRU data in somewhere, and BMC firmware uses ipmi command to get this FRU data.
According to the PXE boot, maybe you could try "Set System Boot Options" command to force system to pxe boot in next reboot.
**Force PXE boot to next boot only:**
```
$ ipmitool raw 0x00 0x08 0x05 0x04 0x00 0x00 0x00
$ ipmitool chassis bootdev pxe
$ ipmitool chassis bootparam set bootflag force_pxe
```

rustyscottweber, 2018-06-06:
I appreciate the traction, however setting the machine to pxeboot is not the issue. The issue is identifying the machine to tell it what to install after it pxe boots.  The system UUID is the best option for me since my systems are all using many many different nics which makes MAC addresses unreliable.

Sigh.. It's never easy is it.

chiabinglee, 2018-06-06:
If your machine has own BMC chip, maybe you could use the Device GUID as reference.

AlexanderAmelkin, 2018-06-06:
Most probably, there is no easy solution to your problem. What most people usually do not realize is that BMC is quite a passive device. It can not fetch arbitrary data or interfere with host system operation in an arbitrary way. It can only provide two types of info:

1. Directly available info, such as:
    * Values/state of directly attached (to BMC chip) sensors
    * FRU information from directly attached (to BMC chip) EEPROMs
    * Information regarding the BMC itself (`ipmitool mc` set of commands)
    * SEL records regarding BMC-related events
2. Information pushed into the BMC by the host or other agents:
    * Values/state of external (to BMC) sensors
    * FRU information from external (to BMC) sources (such as, e.g, mainboard VPD on some OpenPOWER systems)
    * SEL records regarding host boot process and other system events
    * Any other information that the host decides to push into BMC using standard or OEM commands

What you want is that last item in the list. Unless your BIOS pushes that data into BMC using an OEM command and there is a known to you OEM command to fetch that info from BMC, there is no way to know what you want.

rustyscottweber, 2018-06-06:
@chiabinglee , If I use the BMC's guid as a refference, then I have to be able to get the BMC's GUID from ipxe after it boots, which is getting the system UUID at present. It quickly turns into a case of which poison to pick.

@AlexanderAmelkin , It sounds like it should be acceptable to get the vendor of the motherboard using a script and then script out getting the UUID using the vendor specific command based on the vendor.
I think it would be woth it to ask each of the major server vendors to provide instructions on how to get the UUID using ipmitool.  I'll ask them and then post back results here about how to script this out.

AlexanderAmelkin, 2018-06-07:
Sure. You can even ask them to provide patches to ipmitool to their respectime `oem` sections, if they please. I'll keep this ticket closed unless any information comes that would require actions on ipmitool side. However, I do receive notification on all comments, so feel free to update this issue with any new knowledge you gain on this topic.