ipmitool/ipmitool #17: "SOL session closed by BMC" issue

rl0803, 2018-06-27:
We recently hit issue with ipmitool 1.8.18, the SOL intermittently closed with "SOL session closed by BMC" message. Our test result show that BMC is functioning. 

My experiment show that this issue is caused by the 1.8.18 fix ID:449

https://github.com/ipmitool/ipmitool/commit/a3bec1d3658c67f366e1f584abcb5d7a4b5fdce6#diff-9fa8a41d101f194a92375b3b238001ca

Sometimes, the BMC "get device id" responding fall to here to trigger recv_sol(), it expect a SOL message so that a NULL will return and then set the bBmcClosedSession flag. 

The original design fall through this kind of packet, but this fix see it as a BMC close session condition. 
Any good solution to fix this issue?

AlexanderAmelkin, 2018-06-27:
Thank you for reporting. WIll investigate when I have time. If in the meantime you come up with a good fix, feel free to make a pull request.

andreiw, 2019-01-26:
This has been proposed as a fix:

[image]

andreiw, 2019-01-26:
Not my fix and I don't want to delve into the gory details, but this is a long standing bug that's been hitting hard across a number of OpenBMC-based solutions, making it very tedious, as a user, to consume some of the newer servers hitting the market.

iverhan, 2019-04-18:
i met the same issue, recv_sol() will return null cause ipmitool set the bBmcClosedSession flag.
I notice that in ipmi_sol_red_pill()  before recv_sol(), FD_ISSET(intf->fd, &read_fds) is ture ,but in  recv_sol(),the FD_ISSET(intf->fd, &read_fds) become false and make recv-sol return null. I wonder which part of code  close the intf->fd or do something to set the read_fds unreadable.
and do we have some plan to fix this issue in later version. thanks very much

cyang1122, 2019-07-11:
Anyone plan to fix this issue soon?  We see this issue during testing.

psvastkar, 2021-03-09:
In recent testing, I found that my SOL connection is more stable on version 1.8.11 but NOT on 1.8.18. So started looking what is the difference between the both version. I found that the **_select_** function timeout is 5 sec in version 1.8.11 and 1 sec in 1.8.18.
I looked into my tcpdump log and then saw sometime my SOL data packets(response) delayed by 3-5 sec. In that case the **_select_** function is not waiting in version 1.8.18 so it return NULL, so the **_ipmi_lanplus_recv_sol_**(_recv_sol_) function and **_bBmcClosedSession_** flag set to TRUE in **_ipmi_sol_red_pill_** and the local ipmitool session close the sol port. Even the BMC is keep sending the packet the local session is not receiving it as SOL port is already closed. 

​​​​​​​For this issue in v1.8.18, I have added a patch in ipmitool code to handle this issue and you could make use of it.

Here is proposed fix:

`struct ipmi_rs *
ipmi_lan_recv_packet(struct ipmi_intf * intf)
{
	static struct ipmi_rs rsp;
	fd_set read_set, err_set;
	struct timeval tmout;
	int ret;

	FD_ZERO(&read_set);
	FD_SET(intf->fd, &read_set);

	FD_ZERO(&err_set);
	FD_SET(intf->fd, &err_set);
	
	//Added code to fix the issue 17
	if(5 <= intf->session->timeout) {
		tmout.tv_sec = intf->session->timeout;
	}else {
		tmout.tv_sec = 5;
	}
	tmout.tv_usec = 0;

	ret = select(intf->fd + 1, &read_set, NULL, &err_set, &tmout);
	if (ret < 0 || FD_ISSET(intf->fd, &err_set) || !FD_ISSET(intf->fd, &read_set))
		return NULL;`