ipmitool/ipmitool #18: Support Dell iDRAC9 (14G) NIC mode selection

kcgthb, 2018-06-28:
Hi!

It looks like things changed a bit with the latest Dell iDRAC9 generation on 14G servers in terms of selecting the NIC mode (dedicated, shared...), as the `ipmitool delloem lan` commands don't seem to work anymore:
```
# ipmitool delloem lan get

# ipmitool delloem lan set dedicated
Error in setting nic selection (Unknown (0x80))
```

Versions:
```
# ipmitool -V
ipmitool version 1.8.18
# dmidecode -s system-product-name
PowerEdge C6420
# dmidecode -s bios-version
1.4.5
```

I've contacted Dell to get some more info about the change and possibly the IPMI raw commands associated, but I was wondering if anybody else had done this already, so the 14G servers could also be supported by `ipmitool`.

Thanks!

AlexanderAmelkin, 2018-06-29:
@kcgthb, thanks for reporting.
@pwvancil, sorry for bugging, but do you have any ideas on this or know anyone else at Dell who has?

kcgthb, 2018-07-02:
FWIW, it looks like Dell has working patches, because the `ipmitool` version they provide in their [Dell Systems Management Tools and Documentation DVD ISO, v.9.1.0](https://www.dell.com/support/home/us/en/04/drivers/driversdetails?driverId=NP4JN) works on iDRAC9:

```
# ipmitool delloem lan get
Shared LOM   :  shared with lom1
Failover LOM :  None
# ipmitool -V
ipmitool version 1.8.14.dell47

# dmidecode -s system-product-name
PowerEdge C6420
```

I've asked them if they could provide the source modifications, but if anybody has a direct contact w/ engineering, that could be useful too.

ravisaripalli, 2018-07-05:
hi kcgthb

Yes Dell version of ipmi tool will have all the patches, we are trying to post all 14G development patch to open source. 
We already posted some patch already in sourceforge site but neither any of patch got approval or nor any comments from developers.

So now we are planning to post it again.

Thanks
RaviSastry

AlexanderAmelkin, 2018-07-05:
@ravisaripalli, the project has been clinically dead during the past year. I have recently picked it up and moved it here. If you can point me to the mentioned patches on SourceForge, I'll take a look at them. If you have newer versions of the patches, you're more than welcome to make pull requests out of them.

wtip, 2018-09-11:
The ipmitool version in Dell's Ubuntu Xenial repo supports the 14g servers 
https://linux.dell.com/repo/community/openmanage/

kcgthb, 2019-03-04:
Hi @ravisaripalli 
Would you mind maybe re-posting those patches here, instead of on Sourceforge?
Thanks!

patrick-mclean, 2019-03-12:
Is there source for these patches available anywhere?

fdd-rev, 2022-02-18:
If you, anonymous internet user, are in a pinch you can get current setting for 14G with:
```
# ipmitool raw 0x30 0x29
 01 00
```
First byte: 01 = dedicated, 02...05 = shared with lom1...4
Second byte: 00 = no failover, 02...05 = failover on lom1...4, , 06 = failover on all loms

To set:
```
# ipmitool raw 0x30 0x28 0xAA 0xBB
```
Where AA and BB are the first and second byte described for `get`, i.e. LOM selection and second failover selection.

I don't blame the maintainer for not merging the PR as I wouldn't merge it myself. It should be vastly simplified, as the functionality is really, really simple.