ipmitool/ipmitool #19: Board Mfg Date is getting printed in local time making people get confused

ngorugan, 2018-07-09:
The ipmitool is calculation Board Mfg Date as given below(https://github.com/scottjg/ipmitool/blob/master/lib/ipmi_fru.c#L1071)
```
    tval=((fru_data[i+2] << 16) + (fru_data[i+1] << 8) + (fru_data[i]));
    tval=tval * 60;
    tval=tval + secs_from_1970_1996;
    printf(" Board Mfg Date        : %s", asctime(localtime(&tval)));
```

This is giving different Board Mfg Date&time on different host(x86) machines(which are sitting at different time zones) for the same target(BMC) and making people get confused. Can't we print this in GMT time ?

AlexanderAmelkin, 2018-07-09:
That's quite questionable. On one hand, the time is stored in FRU in UTC. On the other hand it is much more natural to see the time in local time zone. Besides, many users, I believe, are used to this local time zone algorithm. All I can think of is adding an option to display all dates in UTC for those who want it.

If you are willing to provide a patch, you're welcome to do so. Just don't forget to read [How to contribute](https://github.com/ipmitool/ipmitool/wiki#how-to-contribute).

ngorugan, 2018-07-17:
I am working on the fix.I will provide the patch soon.

closed 2018-07-19 by AlexanderAmelkin