ipmitool/ipmitool #20: ipmitool sel elist  LUN Num is nor work

ZhangxiuboInspur, 2018-07-11:
When i create two sensor Fan0&Fan1. Fan0 and Fan1 sensornum is same 0x04. Fan0 sensor lun num is 0. Fan1 sensor lun num is 1.

When Fan1 generate some assertion. I use ipmitool -H Ip -U user -P passwd  sel elist . The sensorname is Fan0.

I Checked the ipmi_sdr.c (ipmitool-1.8.18\lib)ipmi_sdr_find_sdr_bynumtype don't compary the lunnum.Only check the sensor num & sensor type.

AlexanderAmelkin, 2018-07-11:
Sounds like a bug. Will look into it.

chiabinglee, 2018-07-12:
Hi @ZhangxiuboInspur 
May I Know what is your purpose of setting these two sensors with the same sensor number?

Refer to the IPMI spec, the Sensor Number is Unique number identifying the sensor behind the given slave address and LUN. Kindly for your reference.

AlexanderAmelkin, 2018-07-13:
@chiabinglee, exactly! They have two sensors behind **different** LUNs. It's quite logical that they have the same sensor number since they are both fans. Apparently they use two similar units, each with its fan and a BMC. Each board is a separate LUN. Sensor numbers overlap. And that's ok as long as LUNs are different. However, `ipmitool` indeed doesn't respect the LUN when searching for sdr by sensor number.

PengWang-SMARTM, 2021-05-02:
LUN should be part of the sensor identifier equation for all sensor related commands (sensor, sdr, sel, and pef).

On the other hand, sensor name may not be unique as a sensor identifier. Some of the commands uses the sensor name as the identifier to find the sensor. Fixing these command would take a little more efforts to handle the situation.

In addition, I wish there could be a command printing the actual LUN:Sensor Number that would help compose a raw command without parsing through the SDR.