ipmitool/ipmitool #21: Unable to use ipmitool with IPv6

TactfulSeal, 2018-07-16:


AlexanderAmelkin, 2018-07-16:
Sorry, can't tell whether this is a problem in ipmitool or in your Windows Server configuration. I'm a Linux person and I don't have Windows, especially Server, to check this behavior.

You're nonetheless welcome to investigate and create a pull request with a fix if you find this indeed is a problem with iptmitool.

TactfulSeal, 2018-07-16:
Thank you all the same. I'll try to investigate this problem.

Xu-Bin, 2019-01-15:
I got the same problem. can ping and web login with ipv6 but ipmitool fails
PS D:\ipmitool> .\ipmitool.exe -vvv -H 2001::6517 -U root -P ***** -I lanplus power status
Testing {951EB64A-CA63-463E-80C5-11EC728D2F44} interface address: fe80::ace6:e832:1dda:5021%16 scope=16
Testing {01F8A7C9-2F3D-44D3-90B7-BA30DD8F10E6} interface address: 2001::6516 scope=0
Testing {01F8A7C9-2F3D-44D3-90B7-BA30DD8F10E6} interface address: fe80::3c05:b997:8857:21a%14 scope=14
Testing {F67F4CA2-1097-4839-A034-07DADF396F19} interface address: fe80::5126:ef6b:379d:4f3e%12 scope=12
Testing {182AD6C9-0DC6-4576-BD93-C13CDED92E6D} interface address: fe80::5efe:172.16.90.138%13 scope=13
Testing {1C496EB5-4645-46A1-86B1-EB23FB5E8BBD} interface address: fe80::5efe:192.168.100.76%15 scope=15
Testing {8D3AFC62-5D40-43F7-A986-B3D7EB918095} interface address: fe80::5efe:192.168.210.223%17 scope=17
Could not open socket!
Error: Unable to establish IPMI v2 / RMCP+ session

chiabinglee, 2019-01-15:
Hi @Xu-Bin,
I considered that the ipmitool of Windows version is not from this project.

The ipmitool is fine to use IPv6 address on Linux.
```
$ ipmitool -I lanplus -H fe80::2d8:61ff:fe01:10c1 -P admin -U admin sel
SEL Information
Version          : 1.5 (v1.5, v2 compliant)
Entries          : 31
Free Space       : 19890 bytes
Percent Used     : 2%
Last Add Time    : 01/15/2019 13:33:04
Last Del Time    : Not Available
Overflow         : false
Supported Cmds   : 'Delete' 'Partial Add' 'Reserve' 'Get Alloc Info'
# of Alloc Units : 1136
Alloc Unit Size  : 18
# Free Units     : 1105
Largest Free Blk : 1105
Max Record Size  : 1
```
FYI~

Xu-Bin, 2019-01-15:
It could be. I download the 1.8.18 source code from sourceforge and build it in my windows server. the procedure as follows:
1. install cygwin including gcc,make and openssl
2. /configure --enable-intf-lanplus
3. make
4. cp ipmitool-1.8.18/src/.libs/ipmitool.exe ipmitool/
5. cp /bin/cygwin1.dll ipmitool/
6. cp /bin/ cygcrypto-1.0.0.dll ipmitool/
7. cp /bin/cygz.dll ipmitool/
8. cp /bin/cyggcc_s-seh-1.dll ipmitool/

it works for ipv4 but failed for ipv6

AlexanderAmelkin, 2019-01-15:
Please try using the latest source code from master branch.
Anyway, if it fails for Windows, I'm afraid I can't help as I explained above.
Let's keep this open and hope that some Windows developer notices it and creates a fix.

AlexanderAmelkin, 2020-07-03:
Please check the latests sources and use the build instructions provided in the `INSTALL` file.
Please tell me if that help or if the problem with IPv6 still exists for you on Windows.

TactfulSeal, 2020-08-17:
I'm sorry,I have leaved my last company, so I'm unable to check it on windows server 2012/2016.  Hope someone else will give it a test.