ipmitool/ipmitool #29: Change the default cipher suite to C17

tomjoseph83, 2018-08-03:
The default cipher suite for ipmitool is Cipher Suite 3. (RAKP-HMAC-SHA1, HMAC-SHA1-96, AES-CBC-128). With many reports indicating SHA1 is unsafe. Can we change the default cipher suite to 17 (RAKP-HMAC-SHA256,HMAC-SHA256-128, AES-CBC-128 ?

Only concern here is that some of the BMC implementations don't support Cipher suite 17 since it is optional and will have to explicitly specify -C 3. I think this is the right way to push users to use stronger algorithms.

ChlorideCull, 2018-08-07:
In my opinion, if possible, fall back to Cipher Suite 3 if Cipher Suite 17 is not available, or introduce the ability to set configuration by environment variable. It would be a lot of pain for me to change all places where it gets invoked to pass `-C 3`, and I don't think Dell will introduce Cipher Suite 17 to my R710s any time soon :smile:

AlexanderAmelkin, 2018-08-07:
@tomjoseph83, can you think of an automatic way to detect the best supported Cipher Suite? As far as I understand, it's impossible to get the list of supported Cipher Suites without first using one to establish a session. I believe that's a flaw in IPMI specification, but alas...

Defaulting to some not commonly supported cipher suite doesn't look like a good idea to me.

tomjoseph83, 2018-08-07:
@AlexanderAmelkin I think there is a way out, the Get Channel Cipher Suites command "works at any privilege level, can be sent prior to a session being established". ipmitool should fire this command before the session setup and then figure out the best possible cipher suite and use that for session setup. In case it fails fallback to -C 3. The best possible cipher suite is chosen by the order of strength of the algorithms.

tomjoseph83, 2018-08-07:
@ChlorideCull Yeah your concern is valid! I definitely don't intend that  IPMI scripts everywhere  change to -C 3 :-) . But this is more of a way to push the BMC implementers to upgrade their firmware to include the support for the SHA 256 algorithms.

With all the browsers ending support for SHA1 SSL certificates, I think we should be pushing to ensure that SHA 256 algorithms is the default option on ipmitool.

AlexanderAmelkin, 2018-08-07:
@tomjoseph83, you're welcome to prepare a pull request then.

vmauery, 2018-08-13:
I had been thinking about this as well. I just pushed something that does this. https://github.com/vmauery/ipmitool/tree/improvement/29

tomjoseph83, 2018-08-14:
Thanks @vmauery :+1:  ! @AlexanderAmelkin we have something ready already for this.

closed 2018-11-01 by AlexanderAmelkin