ipmitool/ipmitool #30: Add undocumented Dell OEM fan control commands

ChlorideCull, 2018-08-07:
As is well known in /r/homelab on reddit, there are some OEM commands for (at least) the 11th Gen Dell PowerEdge servers which can be used to alter the fan control.

Function | `ipmitool raw` command
-----------|--------------------------------
Enable manual fan control | `ipmitool raw 0x30 0x30 0x01 0x00`
Disable manual fan control | `ipmitool raw 0x30 0x30 0x01 0x01`
Set manual fan speed step | `ipmitool raw 0x30 0x30 0x02 0xff 0x??`

These are technically undocumented and not officially supported, and I'm not sure what the manual fan speed steps actually are. I can test it, on a hunch it adjusts the PWM frequency, so I'll just need to dig out a logic analyzer.

----

I also found [this post](https://www.dell.com/community/PowerEdge-Hardware-General/T130-Fan-Speed-Algorithm/td-p/5052692) which details the commands you can send to disable Third Party Card fan behavior.

Function | `ipmitool raw` command
-----------|--------------------------------
Disable 3rd Party Card fan behavior | `ipmitool raw 0x30 0xce 0x00 0x16 0x05 0x00 0x00 0x00 0x05 0x00 0x01 0x00 0x00 `
Enable 3rd Party Card fan behavior | `ipmitool raw 0x30 0xce 0x00 0x16 0x05 0x00 0x00 0x00 0x05 0x00 0x00 0x00 0x00`
Check 3rd Party Card fan behavior | `ipmitool raw 0x30 0xce 0x01 0x16 0x05 0x00 0x00 0x00`

Can't test these though.

----

Then, of course, it's the matter at hand - these are undocumented and entirely unsupported by Dell, and could possibly damage hardware, even. It doesn't feel right to mix them with the rest of the commands. They should either be put in a separate section, or hidden behind a command-line switch, in my opinion, if you even think they should be included.

AlexanderAmelkin, 2018-08-07:
I don't feel like implementing these myself, and I agree that those are dangerous and must be made (if at all) fool-proof. If anyone is willing to implement that and file a pull request, I'll review it.

ChlorideCull, 2018-08-07:
My C is a bit too rusty to help, and those contribution guidelines look daunting. Will be happy to test any implementation though.

AlexanderAmelkin, 2018-08-07:
@ChlorideCull, ehm... what exactly in contribution guidelines looks daunting? The aren't meant to be so.

Anyway, will leave this open for wandering programmers to stumble at it and maybe implement.

ChlorideCull, 2018-08-07:
@AlexanderAmelkin Some stuff seems superfluous, like naming branches, and having to sign off commits (which doesn't really fulfill a reason), which doesn't seem to be enforced. Other than that, it's mostly the fact that I'm used to K&R style code, so it would be a lot of work for me to get anything done.

crazy-logic, 2020-09-06:
I realise this is old but... 

> Set manual fan speed step	`ipmitool raw 0x30 0x30 0x02 0xff 0x??`
> These are technically undocumented and not officially supported, and I'm not sure what the manual fan speed steps actually are. I can test it, on a hunch it adjusts the PWM frequency, so I'll just need to dig out a logic analyzer.

so I have tested my r710 and can confirm that 0x00 is 1200 rpm, 0x64 is 12360rpm and 0x01 through to 0x64. 0x65 has no effect and error's. 0x0a is 2160rpm and is about as quiet as my dell optiplx7010 and a fridge/freezer.

MonkadelicD, 2021-01-28:
> so I have tested my r710 and can confirm that 0x00 is 1200 rpm, 0x64 is 12360rpm and 0x01 through to 0x64 ramp up as they go up. 0x65 has no effect and error's. 0x0a is 2160rpm and is about as quiet as my dell optiplx7010 and a fridge/freezer.

The hex values correspond to PWM %. 0x64 = 100 in decimal. This works down to the minimum fan speed. This is where it can get dangerous. Some users in reddit posts reported setting the value too low turns the fan off which can lead to heat related failures.
On my Dell R720xd 0x64 is ~17500RPM, 0x1a is ~6000RPM, 0x14 is ~4900-5000RPM, 0x0a is ~3300RPM, 0x05 is ~2500RPM. My fans show a minimum speed for 960RPM which looks like that'd be around 0x01. I don't want to set them to that speed at this time and I don't think anyone else would either.

MeStation, 2022-06-17:
anyone can help me find what address to setting "Max Exhaust Temperature" value? i want the fans not actually is static, because there is sometime high loaded cause cpus on high temp.