ipmitool/ipmitool #31: Channel setaccess not accepting privilege

yannlm1, 2018-08-10:
Dear support,

I'm having an issue for setting up a user with USER privilege, as below:

```
channel setaccess 1 6 privilege=2
user list
6   cosmos           true    false      false      NO ACCESS

channel getaccess 1 6 
Maximum User IDs     : 15
Enabled User IDs     : 2

User ID              : 6
User Name            : cosmos
Fixed Name           : No
Access Available     : call-in / callback
Link Authentication  : disabled
IPMI Messaging       : disabled
Privilege Level      : USER
```

As you can see user list is seeing user cosmos as NO ACCESS privilege, and getaccess with USER privilege.
Then when I try to connect via ipmitool I'm getting:
```
Invalid user name
Error: Unable to establish LAN session.

```
On the web interface I can see the user in channel 0 instead of channel 1

[image]

Then when I click on modify user and force enable, it goes to channel 1 (I enabled the user after setting up the password earlier)

[image]

Finally I can connect. My issue is that I need to add this user on hundred machines, can't do it manually by editing via web interface.

Is it a bug or I miss something?

Thanks for your help

AlexanderAmelkin, 2018-08-13:
Hi. First of all, this is not exactly "Support". It's an open community and I'm just a current maintainer of it. But I'm trying to help.

It looks to me like something is wrong with your BMC implementation.
I can't reproduce this on Lenovo IMM 4.43 and a recent version of ipmitool built from sources:

```
$ src/ipmitool -V
ipmitool version 1.8.18.46.g3db2bf6

$ ipmitool -H bmc -I lanplus -U USERID -P PASSW0RD user set name 6 cosmos

$ ipmitool -H bmc -I lanplus -U USERID -P PASSW0RD user set password 6 p@ssw0rd
Set User Password command successful (user 6)

$ ipmitool -H bmc -I lanplus -U USERID -P PASSW0RD channel setaccess 1 6 privilege=2
Set User Access (channel 1 id 6) successful.

$ src/ipmitool -H bmc -I lanplus -U USERID -P PASSW0RD user enable 6

$ src/ipmitool -H bmc -I lanplus -U USERID -P PASSW0RD user list
ID  Name             Callin  Link Auth  IPMI Msg   Channel Priv Limit
1                    true    false      false      NO ACCESS
2   USERID           true    true       true       ADMINISTRATOR
...
6   cosmos           true    true       true       USER
...

$ src/ipmitool -H bmc -I lanplus -U USERID -P PASSW0RD channel getaccess 1
Maximum User IDs     : 13
Enabled User IDs     : 1
...
User ID              : 6
User Name            : cosmos
Fixed Name           : No
Access Available     : call-in / callback
Link Authentication  : enabled
IPMI Messaging       : enabled
Privilege Level      : USER
Enable Status        : enabled

$ src/ipmitool -H bmc -I lanplus -U cosmos -P p@ssw0rd mc info
Set Session Privilege Level to ADMINISTRATOR failed: Unknown (0x81)
Error: Unable to establish IPMI v2 / RMCP+ session

$ src/ipmitool -H bmc -I lanplus -U cosmos -P p@ssw0rd -L user mc info
Device ID                 : 32
Device Revision           : 0
Firmware Revision         : 4.43
IPMI Version              : 2.0
Manufacturer ID           : 19046
Manufacturer Name         : Lenovo Enterprise Business Group
Product ID                : 1049 (0x0419)
Product Name              : Unknown (0x419)
Device Available          : yes
Provides Device SDRs      : no
Additional Device Support :
    Sensor Device
    SDR Repository Device
    SEL Device
    FRU Inventory Device
    IPMB Event Receiver
    IPMB Event Generator
    Chassis Device
Aux Firmware Rev Info     : 
    0x00
    0x00
    0x39
    0x41
```

AlexanderAmelkin, 2018-08-13:
If you still feel like it's a bug in `ipmitool`, you're free to debug it and submit a pull request.
You haven't provided the steps you used to create the user. Are they the same as I used? If not, does it work for you if you repeat my steps?

yannlm1, 2018-08-20:
Thanks for your email, I sorted it out by renaming an existing user, as I had to push this change for hundreds machines.

closed 2018-08-20 by AlexanderAmelkin