ipmitool/ipmitool #37: Missing Error Check in function MD2_Init()

lc3412, 2018-09-17:
Hi,
Function MD2_Init() returns 1 on success, 0 on error. However , the return value of function MD2_Init() in  ipmitool/src/plugins/lan/auth.c  is not checked. See the following code: 

line: 144
[image]

=============================================================================

We find the return value of this call been checked in openssl project with the version of openssl 1.1.2.
Such as in  openssl/crypto/md2 folder

md2/md2_one.c 
line :  26
Ref : https://github.com/openssl/openssl/blob/master/crypto/md2/md2_one.c

[image]

Chi Li, Zuxing Gu, Jiecheng Wu

AlexanderAmelkin, 2018-09-17:
Does this not-checking result in any specific problems?

Anyway, you may prepare a pull request (or wait till I have time to fix this).

lc3412, 2018-09-18:
I will try my best to understande code and provide PR as soon as possible.