ipmitool/ipmitool #40: OpenSSL 1.1.1 error - lanplus_crypt_impl.c:158:17: error: storage size of ‘ctx’ isn’t known

archenroot, 2018-09-24:
I use **sys-apps/ipmitool-1.8.17-r1** release on Gentoo system latest OpenSSL I get error:
`In file included from lanplus.c:57:0:
../../../include/ipmitool/ipmi_lanp.h:130:2: warning: missing initializer for field ‘size’ of ‘struct lan_param’ [-Wmissing-field-initializers]
  { -1 }
  ^
../../../include/ipmitool/ipmi_lanp.h:92:6: note: ‘size’ declared here
  int size;
      ^~~~
lanplus_crypt_impl.c: In function ‘lanplus_encrypt_aes_cbc_128’:
lanplus_crypt_impl.c:158:17: error: storage size of ‘ctx’ isn’t known
  EVP_CIPHER_CTX ctx;
                 ^~~
lanplus_crypt_impl.c:158:17: warning: unused variable ‘ctx’ [-Wunused-variable]
lanplus_crypt_impl.c: In function ‘lanplus_decrypt_aes_cbc_128’:
lanplus_crypt_impl.c:233:17: error: storage size of ‘ctx’ isn’t known
  EVP_CIPHER_CTX ctx;
                 ^~~
lanplus_crypt_impl.c:233:17: warning: unused variable ‘ctx’ [-Wunused-variable]
In file included from ../../../include/ipmitool/ipmi_sel.h:38:0,
                 from lanplus.c:60:
../../../include/ipmitool/ipmi_sdr.h:312:23: warning: ISO C99 doesn’t support unnamed structs/unions [-Wpedantic]
    } ATTRIBUTE_PACKING;
                       ^
`

Do you plan upgrade?

AlexanderAmelkin, 2018-09-24:
Bug reports related to source code not taken from the current master branch are not accepted, sorry. I have no idea what sources Gentoo maintainers use. The current master is build with C11. Please check if it builds for your. For any problems with `ipmitool` sources distributed with any OS type and flavour, please contact maintainers of that distribution.

kapsh, 2020-10-08:
It does build from master. The main problem is that 1.8.18, last marked release, is oudated for ages (for 4 years more specifically) and can't be installed with current openssl versions. The same error I get with 1.1.1h:
```
lanplus_crypt_impl.c: In function 'lanplus_encrypt_aes_cbc_128':
lanplus_crypt_impl.c:167:17: error: storage size of 'ctx' isn't known
167 |  EVP_CIPHER_CTX ctx;
|                 ^~~
lanplus_crypt_impl.c:167:17: warning: unused variable 'ctx' [-Wunused-variable]
lanplus_crypt_impl.c: In function 'lanplus_decrypt_aes_cbc_128':
lanplus_crypt_impl.c:242:17: error: storage size of 'ctx' isn't known
242 |  EVP_CIPHER_CTX ctx;
|                 ^~~
```

Funny thing is, it builds with libressl. But in any case this issue can be resolved by releasing new stable version.