ipmitool/ipmitool #42: Build fails when curl is missing

yoctozepto, 2018-10-16:
When `curl` is missing, `lib/create_pen_list` creates an unusable include file with the following contents:

```
{ , "" },
```

and the build fails with a syntax error. The build should fail early when `curl` cannot be called and `configure` should check whether `curl` is available.

I can fix it and submit a PR but I am time-constrained at the very moment, hence posting an issue.

Have a good day.

yoctozepto, 2018-10-16:
Additionally `make clean` does not remove generated `lib/ipmi_pen_list.inc.c`.

yoctozepto, 2018-10-16:
Ideally, this should be moved to bootstrap anyway.

AlexanderAmelkin, 2018-10-23:
This is a duplicate of #11.

closed 2018-10-23 by AlexanderAmelkin

hdatma, 2019-03-19:
I have curl, and it fails anyway. 
In my case, the problem is conv. 

```
./create_pen_list test.txt                                                                                                                                                       
Generating IANA PEN list...
iconv: conversion from utf8 unsupported
iconv: try 'iconv -l' to get the list of supported encodings
-=O#-   #      #        #   
```
However, I have iconv and it is utf8 capable: 

```
>iconv --version | head -1
iconv (GNU libiconv 1.15)

>iconv -l | grep -i utf.*8
UTF-8
```
So the problem is at line 48 of create_pen_list, where "iconv -f utf8" should be "iconv -f UTF-8".

AlexanderAmelkin, 2019-06-11:
@hdatma, it will all be addressed in context of #11.