ipmitool/ipmitool #50: Failed to enable PEF Policy ID 1

MATT90651, 2018-11-07:
Hello sir
        I want to add enable a policy aler
1、ipmitool -H 10.253.208.11 -I lanplus -U admin -P admin lan alert set 1 1 ipaddr 10.253.178.139
2、ipmitool -H 10.253.208.11 -I lanplus -U admin -P admin pef policy enable 1
but I get this info :IPMI command failed: Invalid data field in request
Failed to enable PEF Policy ID 1.

what did I miss?

AlexanderAmelkin, 2018-11-07:
Most probably that's a question to your BMC manufacturer. "Invalid data field in request" is the response code (0xCC) that ipmitool got from your BMC. You may try to run `ipmitool -v` (or add more `v`s) and see  the data exchange between ipmitool and BMC. Maybe that could shed some light on the reason of the failure.

MATT90651, 2018-11-08:
Hello sir
 I run
 ipmitool -vvv  -H 10.253.208.11 -U admin -P admin -I lanplus lan alert set 1 1 ipaddr 10.253.178.139
get this error info
......
BUILDING A v2 COMMAND
Local RqAddr 0x20 transit 0:0 target 0x20:0 bridgePossible 1
>> Initialization vector (16 bytes)
 5d 9a 87 b7 3a e3 47 8b 4d e2 7a 9a c0 ff 62 04
authcode input (48 bytes)
 06 c0 7d b2 76 79 05 00 00 00 20 00 5d 9a 87 b7
 3a e3 47 8b 4d e2 7a 9a c0 ff 62 04 49 76 49 90
 e8 ef a0 33 24 32 74 3b 9e 4e 18 e8 ff ff 02 07
authcode output (12 bytes)
 ef 71 29 38 5f ae c1 07 e6 97 2c 9b
<< IPMI Response Session Header
<<   Authtype                : RMCP+
<<   Payload type            : IPMI (0)
<<   Session ID              : 0xa0a2a3a4
<<   Sequence                : 0x00000003
<<   IPMI Msg/Payload Length : 48
<< IPMI Response Message Header
<<   Rq Addr    : 81
<<   NetFn      : 07
<<   Rq LUN     : 0
<<   Rs Addr    : 20
<<   Rq Seq     : 03
<<   Rs Lun     : 0
<<   Command    : 01
<<   Compl Code : 0x00
Iana: 37945
Running Get PICMG Properties my_addr 0x20, transit 0, target 0x20

>> Sending IPMI command payload
>>    netfn   : 0x2c
>>    command : 0x00
>>    data    : 0x00 

BUILDING A v2 COMMAND
Local RqAddr 0x20 transit 0:0 target 0x20:0 bridgePossible 1
>> Initialization vector (16 bytes)
 78 92 31 89 86 0f 1a b4 41 70 a8 ee bd 4d 85 f9
authcode input (48 bytes)
 06 c0 7d b2 76 79 06 00 00 00 20 00 78 92 31 89
 86 0f 1a b4 41 70 a8 ee bd 4d 85 f9 3e 22 48 62
 4a 38 39 f1 d7 a3 bf 8d 0e c0 5d 4e ff ff 02 07
authcode output (12 bytes)
 cb a0 7f e3 a6 eb 33 9f 8d 9e 1f e1
<< IPMI Response Session Header
<<   Authtype                : RMCP+
<<   Payload type            : IPMI (0)
<<   Session ID              : 0xa0a2a3a4
<<   Sequence                : 0x00000004
<<   IPMI Msg/Payload Length : 32
<< IPMI Response Message Header
<<   Rq Addr    : 81
<<   NetFn      : 2d
<<   Rq LUN     : 0
<<   Rs Addr    : 20
<<   Rq Seq     : 04
<<   Rs Lun     : 0
<<   Command    : 00
<<   Compl Code : 0xc1
Error response 0xc1 from Get PICMG Properities
Running Get VSO Capabilities my_addr 0x20, transit 0, target 0x20

>> Sending IPMI command payload
>>    netfn   : 0x2c
>>    command : 0x00
>>    data    : 0x03 

BUILDING A v2 COMMAND
Local RqAddr 0x20 transit 0:0 target 0x20:0 bridgePossible 1
>> Initialization vector (16 bytes)
 6a d4 40 2b 84 f7 5a 22 75 74 58 32 25 46 0f b6
authcode input (48 bytes)
 06 c0 7d b2 76 79 07 00 00 00 20 00 6a d4 40 2b
 84 f7 5a 22 75 74 58 32 25 46 0f b6 ab 86 f2 bb
 e7 1b 03 50 d9 18 d1 d5 ea 77 a3 df ff ff 02 07
authcode output (12 bytes)
 37 04 18 94 70 af 9d 3e 4e 7f 10 c9
<< IPMI Response Session Header
<<   Authtype                : RMCP+
<<   Payload type            : IPMI (0)
<<   Session ID              : 0xa0a2a3a4
<<   Sequence                : 0x00000005
<<   IPMI Msg/Payload Length : 32
<< IPMI Response Message Header
<<   Rq Addr    : 81
<<   NetFn      : 2d
<<   Rq LUN     : 0
<<   Rs Addr    : 20
<<   Rq Seq     : 05
<<   Rs Lun     : 0
<<   Command    : 00
<<   Compl Code : 0xc1
Invalid completion code received: Invalid command
Acquire IPMB address
Discovered IPMB address 0x0
Interface address: my_addr 0x20 transit 0:0 target 0x20:0 ipmb_target 0
......

and 
How do I configure thiss command :
 pef policy create <id = 1..n> <params>

AlexanderAmelkin, 2018-11-08:
What happens here is:
1. ipmitool asks for Device ID.
2. Your BMC reports that it is an Inspur device (37945).
3. ipmitool tries to ask for PICMG properties
4. You BMC responds with an "Invalid command" (0xc1) code indicating that the command is not supported.
5. ipmitool tries to acquire IPMB address
6. You BMC responds with an "Invalid command" (0xc1) code indicating that the command is not supported.

What happens next is unknown as you haven't provided the full log. I don't think those two "invalid command" responses from your BMC were any problem. The part of the log you gave doesn't even get to sending the 'set alert' command. Also, next time please attach the log (as .txt), don't copy-paste it.

MATT90651, 2018-11-09:
this is my 'alert set'  log and 'enable policy' log.
I'm very urgent to solve this problem.
please.
[enable policy.log](https://github.com/ipmitool/ipmitool/files/2565275/enable.policy.log)
[alert set log.log](https://github.com/ipmitool/ipmitool/files/2565276/alert.set.log.log)

AlexanderAmelkin, 2019-02-21:
Sorry for being silent for so long. I've looked into those full logs and I can see:

```
PEF Policy ID 1 is enabled now
```

in [enable policy.log](https://github.com/ipmitool/ipmitool/files/2565275/enable.policy.log),
and in [alert set log.log](https://github.com/ipmitool/ipmitool/files/2565276/alert.set.log.log) I see:

```
>> Sending IPMI command payload
>>    netfn   : 0x0c
>>    command : 0x01
>>    data    : 0x01 0x13 0x01 0x00 0x00 0x0a 0xfd 0xb2 0x8b 0x00 0x00 0x00 0x00 0x00 0x00 
```
This is "Set LAN Configuration parameter, channel 1,  parameter 0x13 (LAN Alert Destination Address), destination 1, IPv4 address, 10.253.178.139, zero MAC-address"

The command succeeds:
```
<< IPMI Response Session Header
...
<<   Compl Code : 0x00
```
Then there is:
```
>> Sending IPMI command payload
>>    netfn   : 0x0c
>>    command : 0x01
>>    data    : 0x01 0x00 0x02 
```
Which is "Set LAN Configuration parameter, channel 1, parameter 0x00 (Set in progress), commit write". The BMC responds that this is not supported:

```
<< IPMI Response Message Header
<<   Compl Code : 0x80
```

But that's ok. Then there is:

```
>> Sending IPMI command payload
>>    netfn   : 0x0c
>>    command : 0x01
>>    data    : 0x01 0x00 0x00 
...
<< IPMI Response Message Header
...
<<   Compl Code : 0x00
```
That's a successful clearing of "Set in progress" flag.

All in all that indicates that everything worked.
Are you still having problems?

I agree though that those `Invalid completion code received: Invalid command` messages are misleading and must not be printed. They just indicate that your BMC is not PICMG/HPM.2 compliant.