ipmitool/ipmitool #53: RAKP 2 message error

YiannisGkoufas, 2018-11-09:
I have troubles connecting to an fsp with ipmitool.
The command I issue is:
```
ipmitool -I lanplus -H xxx.xxx.xxx.xxx -U xxxx -P xxxx chassis power status -vvv
```
And the error I get is:
```
>> SENDING A RAKP 1 MESSAGE

<<RAKP 2 MESSAGE
<<  Message tag                   : 0x00
<<  RMCP+ status                  : illegal parameter
<<  Console Session ID            : 0x00000000
<<  BMC random number             : 0x00000000000000000000000000000000
<<  BMC GUID                      : 0x00000000000000000000000000000000
<<  Key exchange auth code [sha1] : 0xffff020700000000000000000000000000000000

RAKP 2 message indicates an error : illegal parameter
Error: Unable to establish IPMI v2 / RMCP+ session
```

Not sure if its an authorization issue or something else.
Any help would be highly appreciated.
Thanks!

AlexanderAmelkin, 2018-11-09:
Are you sure you need `-I lanplus` ? Try `-I lan`. Looks like some error reported by your BMC. Probably it just doesn't support RMCP+.

YiannisGkoufas, 2018-11-09:
Hi @AlexanderAmelkin thanks for the feedback, now I got:
```
Authentication type NONE not supported
```
This means wrong username/password?

AlexanderAmelkin, 2018-11-11:
No, it actually means that you indeed need `lanplus`. Please attach (don't copy-paste) the full log and please run `ipmitool -vvvvvv` (that is, 5+ `-v` options). That will ensure the maximum verbosity level.

YiannisGkoufas, 2018-11-15:
It must have been an authorization issue, because I managed to access the ipmitool on the node in question and set the user/pass and after that it worked.
Really appreciate that you followed up, thanks again for your time!

closed 2018-11-22 by YiannisGkoufas

YiannisGkoufas, 2018-11-22:
In case someone runs into the issue, the way to fix it is the section **User Configuration** from here https://www.thomas-krenn.com/en/wiki/Configuring_IPMI_under_Linux_using_ipmitool
Basically you need to configure the admin user as follows:
```
[root@sr2500 ~]# ipmitool user set name 2 admin
[root@sr2500 ~]# ipmitool user set password 2
Password for user 2: 
Password for user 2: 
[root@sr2500 ~]# ipmitool channel setaccess 1 2 link=on ipmi=on callin=on privilege=4
[root@sr2500 ~]# ipmitool user enable 2
```