ipmitool/ipmitool #54: ipmitool --help seems to print to stderr

veita, 2018-11-10:
`ipmitool --help` seems to print to stderr. This makes it cumbersome to read the full help text on the terminal. `ipmitool --help | less` doesn't work as expected.

See https://www.gnu.org/prep/standards/html_node/_002d_002dhelp.html

Affected version: `ipmitool version 1.8.17 on Gentoo Linux`

AlexanderAmelkin, 2018-11-11:
ipmitool is not a GNU program. There are other tools as well that print help to stderr. You may use `ipmitool help 2>&1 | less`.

veita, 2018-11-11:
OK, thank you, I see.

Just out of curiosity: Are there use-cases where printing `--help` to stderr has advantages over printing to stdout?

AlexanderAmelkin, 2018-11-12:
I can only think of not getting to parse help output when you've passed some wrong arguments in a script. This discussion seconds that use case: https://stackoverflow.com/questions/2199624/should-the-command-line-usage-be-printed-on-stdout-or-stderr

I know though that ipmitool always prints it to stderr, but it's been like that forever and I don't see any strong reason to change that.

veita, 2018-11-12:
Alexander, thank you for your reply. So there's no single standard how to handle this issue. Personally, I would prefer a more convenient command to read `--help` output, but I'll keep your redirect solution in mind. ;-)

AlexanderAmelkin, 2018-11-14:
You're welcome. I too find this inconvenient, but that's how ipmitool traditionally behaved and I don't want to break that without a better reason than my or your feeling.

closed 2018-11-14 by AlexanderAmelkin