ipmitool/ipmitool #57: IPMI timestamp definition are different from Mfg Date of FRU's definition.

AndreaYC, 2018-11-14:
It seems to me the issue still exist because of different definition we had reference.
According to "Platform Management FRU Information Storage Definition
v1.0 rev 1.3", Table 11-1 
[image]
 IPMI timestamp definition are different from FRU definition.
[image]

Maybe we still need to add a determine statements to check the unspecified information of Board Mfg. Date/Time field?

_Originally posted by @AndreaYC in https://github.com/ipmitool/ipmitool/issues/51#issuecomment-438158795_

AlexanderAmelkin, 2018-11-14:
Please demonstrate with ipmitool output how exactly the issue still exists for you.

From my point of view, exactly this issue has been addressed by:
```c
       if (!ipmi_timestamp_is_valid(stamp)) {
               return "Unspecified";
       }
```
in `lib/ipmi_time.c`.

Also, there is no need to create more duplicate issues. You could have commented in #51 if you believed it was not fixed by #23.

AlexanderAmelkin, 2018-11-14:
If you have an example of a FRU at which `ipmitool` fails to display the correct date, please gather it with `ipmitool fru read` and attach here for reference.