ipmitool/ipmitool #79: Provide helper macro for freeing pointers

pstrinkle, 2018-11-29:
Either:

#define FREE(PTR) do { free(PTR); (PTR) = NULL; } while(0);

or

void free_and_nullify(void **ptr) {
    if (ptr && *ptr) {
        free(*ptr);
        *ptr = NULL;
    }
}

This will save developers from needing to set a pointer explicitly to null in cases where doing so is non-intuitive, e.g. at the end of a method.

AlexanderAmelkin, 2018-12-05:
I like the idea of a function better, because there is less chance to misuse it.
The macro you add with #80 can hypothetically be called as `FREE(pptr++)` and will result in pptr being checked for NULL, but then `free()` called on `pptr+1`. I don't know why would anyone do it, but I generally don't know why people do stupid things. So let's just prevent that with an **inline** function.

Let's call it something shorter though. Like `zfree()` for example.

pstrinkle, 2018-12-05:
Amusingly this started as an inline method but using it required a bit more
tweaking to the code.

pstrinkle, 2018-12-05:
zfree implies it's going to zero the memory before it's freeing it.

void free_and_nullify(void **ptr) { if (ptr && *ptr) { free(*ptr); *ptr = NULL; } } could be, nfree or nullfree -- which sounds weird.

pstrinkle, 2018-12-05:
free_n() works, it'll be up for review momentarily.  In case you're curious, I've carved out a lot of cycles to get these PRs merged and fix everything we can find.

PR #83 (2018-12-24): helper: add free_n method to handle clearing pointers
free_n() will free the memory and clear the pointer, which will reduce
the probability a developer will forget to clear the pointer after
freeing.
Resolves: #79
commit 8b6d127b (2018-12-24): helper: add free_n method to handle clearing pointers

free_n() will free the memory and clear the pointer, which will reduce
the probability a developer will forget to clear the pointer after
freeing.

Resolves: #79
closed 2018-12-24 by AlexanderAmelkin