ipmitool/ipmitool #84: nm functionality fails on recent intel servers (S2600BPB skylake xeon)

nsccap, 2018-12-18:
nm commands fail with Unknown error. Example: "nm statistics power":
    NM request failed because: Unknown (0xC1) (c1)

Other tested commands with same error: "discover", "capabilities"

Corresponding nm data is confirmed available over https on the server.

Versions of ipmitool tested: rhel7 1.8.18 and (as of today current git) 34fc86b

Tested both on lanplus and local.

Test server responds correctly for "normal" ipmi commands.

Multiple servers tested with identical behavior.

Server firmware details:
BMC Firmware Build Time :	Fri May 4 04:07:41 2018
BIOS ID :	SE5C620.86B.00.01.0013.030920180427
BMC FW Rev :	1.43.91f76955
Backup BMC FW Rev :	1.43.660a4315
Build ID :	91F76955
SDR Package Version :	1.30
Mgmt Engine (ME) FW Rev :	04.00.04.294

AlexanderAmelkin, 2018-12-19:
Are you sure that your BMC has NodeManager commands support?
0xC1 is actually a standard IPMI response code for "Invalid command" that usually means that the requested command is simply not supported by the BMC.

Have this ever worked for you on any of your servers? Does it work with other tools like `ipmiutil`?
Why do you think it should work on your hardware?

Unfortunately, I don't have any hardware at hand with support for Intel NodeManager, so I just can't verify.

Please also attach a full log of `ipmitool -vvvvv nm capabilities` so that I could see what's happening there.

AlexanderAmelkin, 2018-12-19:
@vmauery, can you please check if `nm` functionality in `ipmitool` is working for you?

chiabinglee, 2018-12-20:
Hi @nsccap,
Kindly use the IPMI bridge raw command or "mc info" command to check if NM workable.
Such as,
```
ipmitool -b 0x06 -t 0x2c raw 0x06 0x01
ipmitool -b 0x06 -t 0x2c mc info
```
The device info will response from NM.

The commands at below page is for your reference:
[https://computercheese.blogspot.com/2013/06/ipnm-ipmi-commands-supported-in.html](url)

nsccap, 2018-12-20:
> Are you sure that your BMC has NodeManager commands support?

Yes

> 0xC1 is actually a standard IPMI response code for "Invalid command" that usually means that the requested command is simply not supported by the BMC.
>
> Have this ever worked for you on any of your [...]

I did, after ipmitool failed me, go on a hunt for something that works. It turns out that freeipmi-1.6.2
works ok and can get all the expected nm data including for example "statistics power".
 
> Unfortunately, I don't have any hardware at hand with support for Intel NodeManager, so I just can't verify.
> 
> Please also attach a full log of `ipmitool -vvvvv nm capabilities` so that I could see what's happening there.

I will add this but it may take a while (just left $work for the holidays)

Thanks for jumping on this

nsccap, 2018-12-20:
> Hi @nsccap,
> Kindly use the IPMI bridge raw command or "mc info" command to check if NM workable.
> Such as,

I will try these commands but it may take some time (just left $work for the holidays).
 
> ```
> ipmitool -b 0x06 -t 0x2c raw 0x06 0x01
> ipmitool -b 0x06 -t 0x2c mc info

> The commands at below page is for your reference:
> https://computercheese.blogspot.com/2013/06/ipnm-ipmi-commands-supported-in.html

The pdf linked on the page you linked refers to NM 2.0. I think now 5 years on it's at NM 4.

PuwenHsiao, 2018-12-31:
> nm commands fail with Unknown error. Example: "nm statistics power":
> NM request failed because: Unknown (0xC1) (c1)
>
> Other tested commands with same error: "discover", "capabilities"
>
> Corresponding [...]

Hi @nsccap ,
Have you check then channel / address of NM?
For example, I have a Supermicro Server and its Channel No is 0x00 and Address is 0x2C.
I need to use this information in ipmitool. Otherwise I will receive an error message.

$ sudo ipmitool -b 0x00 -t 0x2c nm statistics power
    Power domain:                             platform
    Policy/Global Admin state                 Globally Enabled
    Policy/Global Operational state           suspended
    Policy/Global Measurement state           in progress
    Policy Activation state                   not triggered
    Instantaneous reading:                         184 Watts
    Minimum during sampling period:                176 Watts
    Maximum during sampling period:                246 Watts
    Average reading over sample period:            193 Watts
    IPMI timestamp:                           Mon Dec 31 14:07:13 2018
    Sampling period:                          04471029 Seconds.

$ sudo ipmitool nm statistics power
    NM request failed because: Unknown (0xC1) (c1)

AlexanderAmelkin, 2019-01-10:
The default is to not bridge. In your system the NM is a separate BMC, not the main one. That separate BMC is apparently connected to the i2c bus 6 of the main BMC. That is the kind of information you get from your system vendor. `ipmitool` just implements the NM command set and has no idea of which BMC unit implements the actual functionality.

closed 2019-01-10 by AlexanderAmelkin

nsccap, 2019-01-10:
Sounds like you consider this expected behavior then. Please feel free to close the issue.

We've added the needed bridge and target numbers to our local wrapper. Thanks for taking time to help me out.

However I do note that freeipmi manages as long as it's told it's intel nm (meaning either lucky hard code, brute force or a method of discovery):

# ipmi-oem Intelnm get-node-manager-statistics mode=globalpower
Current Power                                 : 149 Watts
Minimum Power                                 : 1 Watts
Maximum Power                                 : 426 Watts
...