ipmitool/ipmitool #85: On Dell server observing the "Unable to send command: Invalid argument" when issued sdr list command

ravisaripalli, 2018-12-20:
When issuing sdr list command on DELL servers, we are observing the "Unable to send command: Invalid argument".  This error can be seen for sdr which are not a BMC owned sensor and sensor owner ID is not 0x20.

AlexanderAmelkin, 2019-02-21:
Can you please run `ipmitool -vvvvvvvvv sdr list` (that is, with nine or more `-v`) and attach the log here?

ravisaripalli, 2019-03-04:
[ipmi_sdr_output.txt](https://github.com/ipmitool/ipmitool/files/2925383/ipmi_sdr_output.txt)
@AlexanderAmelkin Added the verbose out put with -v option (as suggested). From the output, error will be seen when the target address is 0xb1,
" Sending request 0x2d to IPMB target @ 0xb1:0x0 (from 0x20)
Unable to send command: Invalid argument"

As per the IPMI Specification 2.0 the target address can't be odd value, so ipmi tool should restrict any such address. 
[image]
[image]

AlexanderAmelkin, 2019-03-04:
Looks like a bug indeed. SDRs that have `ID[0] = 1` must not be bridged, but it appears that they are. Bridging support is quite poor in `ipmitool` indeed. I'll keep this ticket open and will work on it when I have time.

AlexanderAmelkin, 2019-03-04:
This is also related, although not directly, to #20.

ravisaripalli, 2019-03-04:
The below pull request is the fix for the above bug. 
https://github.com/ipmitool/ipmitool/pull/86

But again due to wrong commit ID and lack of git knowledge it went wrong way.
I can squesh my pervioues commit and repost changes includeing code review comments.

AlexanderAmelkin, 2019-03-05:
Oh, I didn't realize pull #86 was related. Please see my latest comments there. I don't think the fix is correct.

Jean-Daniel, 2019-03-08:
Just to let you know I just got the same error running `ipmitool sensor` (in case it does not use the same code path, and need special care in the patch too).

ravisaripalli, 2019-03-20:
But in the fix i am just trying to block the Sensor owner id only. And allowing all other address.

PengWang-SMARTM, 2021-05-02:
@ravisaripalli This is an interesting question. Are there any updates?

To my understanding, since the owner ID is a software ID, the command should actually be bridged to the system interface, not "no bridge".

Are you able to compose a raw "send message" command to read this sensor?