ipmitool/ipmitool #89: IPMI : "dcmi power get_limit" command does not work via inband IPMI

rahulmah, 2019-01-02:
"dcmi power get_limit" command does not work via inabnd IPMI. But with out-of-band IPMI, it is working fine.

```
[root@ws006p1 ~]# ipmitool dcmi power get_limit

    A valid DCMI command was not returned! (80)
[root@ws006p1 ~]#
```
```
bash-4.1$ ipmitool -I lanplus -C 3 -P 0penBmc -H x.x.x.x dcmi power get_limit

    Current Limit State: No Active Power Limit
    Exception actions:   Hard Power Off & Log Event to SEL
    Power Limit:         3000 Watts
    Correction time:     0 milliseconds
    Sampling period:     0 seconds
```

The same completion code is returned for both network IPMI and inband IPMI. 0x80 stands for no active power limit set.

```
<< IPMI Response Message Header
<<   Rq Addr    : 81
<<   NetFn      : 2d
<<   Rq LUN     : 0
<<   Rs Addr    : 20
<<   Rq Seq     : 06
<<   Rs Lun     : 0
<<   Command    : 03
<<   Compl Code : 0x80
```
The network IPMI interprets the completion code correctly as "No Active Power Limit" in the case of network IPMI and in the case of inband IPMI the return code is not interpreted correctly.

AlexanderAmelkin, 2019-02-21:
Funny. Will look into it.

galshwed, 2019-05-12:
Is there any update regarding this bug?
Currently working on a new product that uses ipmitool and its power capping capabilities in particular.
Please let me know if I can help in any way.
I would be glad if I can be updated in galshw@mellanox.com.

doublechiang, 2020-02-19:
I tested again with most updated ipmitool and can't find this symptom.
`
$ sudo ipmitool -V
ipmitool version 1.8.18.179.gc3939da
$ sudo ipmitool dcmi power get_limit

    Current Limit State: Power Limit Active
    Exception actions:   Hard Power Off & Log Event to SEL
    Power Limit:         500 Watts
    Correction time:     1000 milliseconds
    Sampling period:     5 seconds
`

op47147, 2022-07-12:
Try to activate first.
```
ipmitool dcmi power activate
ipmitool dcmi power get_limit
```