ipmitool/ipmitool #111: Ipmitool sends incorrect format of Platform Event when it's using SSIF (SMBUS System Interface)

vlognouh, 2019-03-15:
Hi,
As item "29.6 - System Interface Event Request Message Format" in IPMI spec v2 (Document Revision 1.1, October 1, 2013) mentioned that SSIF (SMBUS System Interface) uses the same format as KCS. It means that the Platform Event message sent via SSIF should have an extra Generator ID byte like KCS but the current code did not check Channel Medium Type is SMBUS to add this byte into the message. Currently, it only adds this extra byte when Channel Medium Type is 0x0C which means KCS, BT or SMIC. Please check below code in lib/ipmi_event.c:

503 / setup Platform Event Message command /
504 memset(&req, 0, sizeof(req));
505 req.msg.netfn = IPMI_NETFN_SE;
506 req.msg.cmd = 0x02;
507 req.msg.data = rqdata;
508 req.msg.data_len = 7;
509
510 chmed = ipmi_current_channel_medium(intf);
511 if (chmed == IPMI_CHANNEL_MEDIUM_SYSTEM) {
512 / system interface, need extra generator ID /
513 rqdata[0] = 0x41; // As per Fig. 29-2 and Table 5-4
514 req.msg.data_len = 8;
515 }

AlexanderAmelkin, 2019-03-15:
Thanks for reporting this.
Do you have hardware to verify this once it is fixed?
Are you by any chance willing to make a pull request with a fix?

adarshgrunni, 2019-05-20:
I have added the fix in https://github.com/ipmitool/ipmitool/pull/125#issue-280387205.
The same is verified in hardware with ssif .

AlexanderAmelkin, 2019-06-10:
@vlognouh, please see/join the discussion in #125. What is the firmware that your BMC is using?

AlexanderAmelkin, 2019-07-01:
@vlognouh, @adarshgrunni, are you still interested in a proper fix for this? Please see and comment #138.

PR #138 (2019-07-02): Bugfix/111 missing generator id on ssif si
This fixes #111
closed 2019-07-02 by AlexanderAmelkin