ipmitool/ipmitool #122: Error: Unable to establish IPMI v2 / RMCP+ session

moonshineBoy, 2019-04-26:
```
[ldl@localhost ~]$ ipmitool -H 172.18.16.230 -I lanplus -U ADMIN  power status
Password: 
Error: Received an Unexpected Open Session Response
Error: Received an Unexpected Open Session Response
Error: Received an Unexpected Open Session Response
Error: Received an Unexpected Open Session Response
Error: Received an Unexpected Open Session Response
Error: Unable to establish IPMI v2 / RMCP+ session

```

when I ping the ip i get bellow output:

```
[ldl@localhost ~]$ ping 172.18.16.230
PING 172.18.16.230 (172.18.16.230) 56(84) bytes of data.
From 172.18.31.252 icmp_seq=1 Redirect NetworkFrom 172.18.31.252: icmp_seq=1 Redirect Network(New nexthop: 172.18.16.230)
64 bytes from 172.18.16.230: icmp_seq=1 ttl=64 time=16.2 ms
64 bytes from 172.18.16.230: icmp_seq=1 ttl=64 time=16.2 ms (DUP!)
64 bytes from 172.18.16.230: icmp_seq=1 ttl=63 time=16.2 ms (DUP!)
64 bytes from 172.18.16.230: icmp_seq=1 ttl=63 time=16.2 ms (DUP!)
64 bytes from 172.18.16.230: icmp_seq=1 ttl=62 time=16.3 ms (DUP!)
64 bytes from 172.18.16.230: icmp_seq=1 ttl=62 time=16.3 ms (DUP!)
From 172.18.31.252 icmp_seq=2 Redirect NetworkFrom 172.18.31.252: icmp_seq=2 Redirect Network(New nexthop: 172.18.16.230)
64 bytes from 172.18.16.230: icmp_seq=2 ttl=64 time=0.946 ms
64 bytes from 172.18.16.230: icmp_seq=2 ttl=64 time=0.967 ms (DUP!)
64 bytes from 172.18.16.230: icmp_seq=2 ttl=63 time=1.19 ms (DUP!)
64 bytes from 172.18.16.230: icmp_seq=2 ttl=63 time=1.20 ms (DUP!)
64 bytes from 172.18.16.230: icmp_seq=2 ttl=62 time=1.20 ms (DUP!)
64 bytes from 172.18.16.230: icmp_seq=2 ttl=62 time=1.20 ms (DUP!)
From 172.18.31.252 icmp_seq=3 Redirect NetworkFrom 172.18.31.252: icmp_seq=3 Redirect Network(New nexthop: 172.18.16.230)
64 bytes from 172.18.16.230: icmp_seq=3 ttl=64 time=1.05 ms
64 bytes from 172.18.16.230: icmp_seq=3 ttl=64 time=1.07 ms (DUP!)
64 bytes from 172.18.16.230: icmp_seq=3 ttl=63 time=1.07 ms (DUP!)
64 bytes from 172.18.16.230: icmp_seq=3 ttl=63 time=1.07 ms (DUP!)
64 bytes from 172.18.16.230: icmp_seq=3 ttl=62 time=1.30 ms (DUP!)
64 bytes from 172.18.16.230: icmp_seq=3 ttl=62 time=1.30 ms (DUP!)
^C
--- 172.18.16.230 ping statistics ---
3 packets transmitted, 3 received, +15 duplicates, +3 errors, 0% packet loss, time 2004ms
rtt min/avg/max/mdev = 0.946/6.186/16.304/7.146 ms
[ldl@localhost ~]$ 

```

moonshineBoy, 2019-04-29:
the detail I post on the SO: https://superuser.com/questions/1430642/after-several-times-of-ipmitool-commands-now-i-can-not-get-established-with-ipm

AlexanderAmelkin, 2019-04-29:
You obviously have problems with your network setup. Looks like you are routing packets to a directly reachable destination. As you can see, you get duplicate ICMP responses even with `ping`. I'm quite confident you problem has nothing to do with `ipmitool` as such.

closed 2019-04-29 by AlexanderAmelkin