ipmitool/ipmitool #124: Vendor Newisys changed to Viking Enterprise

danm3986, 2019-05-15:
Nov. 12, 2018 – Newisys, a product division of Sanmina Corporation (Nasdaq: SANM) has been renamed to Viking Enterprise Solutions

danm3986, 2019-05-15:
[bugfix-124.zip](https://github.com/ipmitool/ipmitool/files/3183587/bugfix-124.zip)

AlexanderAmelkin, 2019-05-31:
Can you please create a pull request instead? That way it will automatically undergo CI testing.

danm3986, 2019-05-31:
done,thank you

-- 
Dan McGee
Engineer
Newisys - A Division of Sanmina Corporation
5385 Mark Dabling Blvd., Suite 100
Colorado Springs, CO  80918
*Cell - 719 243 9353*
dan.mcgee@sanmina.com <randi.suttles@sanmina.com>

-- 
CONFIDENTIALITY
This e-mail message and any attachments thereto, is 
intended only for use by the addressee(s) named herein and may contain 
legally privileged and/or confidential information. If you are not the 
intended recipient of this e-mail message, you are hereby notified that any 
dissemination, distribution or copying of this e-mail message, and any 
attachments thereto, is strictly prohibited.  If you have received this 
e-mail message in error, please immediately notify the sender and 
permanently delete the original and any copies of this email and any prints 
thereof.
ABSENT AN EXPRESS STATEMENT TO THE CONTRARY HEREINABOVE, THIS 
E-MAIL IS NOT INTENDED AS A SUBSTITUTE FOR A WRITING.  Notwithstanding the 
Uniform Electronic Transactions Act or the applicability of any other law 
of similar substance and effect, absent an express statement to the 
contrary hereinabove, this e-mail message its contents, and any attachments 
hereto are not intended to represent an offer or acceptance to enter into a 
contract and are not otherwise intended to bind the sender, Sanmina 
Corporation (or any of its subsidiaries), or any other person or entity.

closed 2019-06-11 by AlexanderAmelkin