ipmitool/ipmitool #126: RAKP 3 is not sent when there is an authentication error in RAKP 2

lohaukai, 2019-05-22:
The issue was discovered when I was testing "Bad Password Threshold" that "-I lanplus" would invalid it out.

After checking the source code, it was result from an early return of  `ipmi_lanplus_rakp1` in **lanplus.c**. 

I remove `rc = 1;` behind `session->v2_data.rakp2_return_code = IPMI_RAKP_STATUS_INVALID_INTEGRITY_CHECK_VALUE;`, and verified that is able to fix the issue.

Is there any known issue, reason or planed fix for this one?

```patch
diff --git a/src/plugins/lanplus/lanplus.c b/src/plugins/lanplus/lanplus.c
old mode 100644
new mode 100755
index 28cb31c..72aba31
--- a/src/plugins/lanplus/lanplus.c
+++ b/src/plugins/lanplus/lanplus.c
@@ -3120,7 +3120,7 @@ ipmi_lanplus_rakp1(struct ipmi_intf * intf)
 			/* Error */
 			lprintf(LOG_INFO, "> RAKP 2 HMAC is invalid");
 			session->v2_data.rakp2_return_code = IPMI_RAKP_STATUS_INVALID_INTEGRITY_CHECK_VALUE;
-								rc = 1;
+								// rc = 1;
 		}
 		else
 		{
```

vmauery, 2019-06-06:
If the authentication in RAKP2 was not correct, there is no need to move on to RAKP3. The code's current behavior is correct.

adarshgrunni, 2019-06-11:
I think if the authentication in RAKP2 was not correct also it should move to  RAKP3. then only the  Bad Password Threshold count and based on that operations  can manage from server side. If it is stopped at RAKP2 client side ,from server side we will not get any info about no of wrong attempts ,etc.

Correct me if i am wrong..

vmauery, 2019-06-11:
The server cannot rely on a client to always send RAKP3 in order to do its password threshold counting. Sure, this might be called good behavior. But a poorly behaved client could always simply just not send RAKP3 if it wants to attack. You are right in that the server would not be able to directly tell if the client has a bad password unless it gets RAKP3.

This is yet another hole in the RMCP+ protocol. Use redfish. :)

lohaukai, 2019-06-13:
@vmauery According to "The server cannot rely on a client", could BMC monitor the timeout, waiting for RAKP3, to count the  failure login request?

vmauery, 2019-06-14:
Yes, the BMC could monitor for timeouts, but that does not necessarily mean that the timeout was caused because of a bad login. The issue here is that for the BMC to know that the login failed, a RAKP3 would need to be sent, but it makes no sense to send a RAKP3 if RAKP2 failed authentication. So really, you can only successful track login failures from helpful clients. An attacker can take advantage of this and simply not send RAKP3 and it would never get logged.

Probably the best deterrent here would be for the BMC to only allow for a single authentication session to be open from any given IP address with a short timeout that could also be canceled by a new session attempt. The built-in assumption here is that if a login fails, it will require a second or two to rectify the password. Then, back-to-back timeouts could be logged as suspicious or even as possibly incorrect logins. Maybe this would not work or there are some holes in it, but the protocol does not really offer a way to solve this correctly.