ipmitool/ipmitool #128: ipmitool shouldn't default to lan intf

wkennington, 2019-06-02:
Without any configuration, ipmitool is now defaulting the lan interface. This is surprising to and end user who is used to being able to do `ipmitool <cmd>` on a machine that has a local ipmi interface. It's not particularly useful to default to the lan interface anyway since it requires additional, explicit configuration.

AlexanderAmelkin, 2019-06-03:
I believe this is because of #117, but it's quite surprising because as you may see the change introduced there takes care of defaulting to 'open' on Linux. It indeed defaults to 'lan' on MacOS. If that is wrong (sorry, I don't have MacOS), then please tell what must be the default interface for it or create a pull request yourself to refix #117.

wkennington, 2019-06-04:
I'd have to dig into it more but it is on linux, and the open interface was detected correctly.

[build.log](https://github.com/ipmitool/ipmitool/files/3249927/build.log)

closed 2019-06-10 by AlexanderAmelkin