ipmitool/ipmitool #131: Buffer overflow in fru_area_print_product

CajuM, 2019-06-06:
Hello,
I've encountered heap corruption while using ipmitool. After debugging I've found that the problem is as follows:
The code at https://github.com/ipmitool/ipmitool/blob/master/lib/ipmi_fru.c#L1178 is:
```
fru_data = malloc(fru_len);
if (read_fru_area(intf, fru, id, offset, fru_len, fru_data) < 0)
```
But the documentation of read_fru_area says that it will write length bytes into frubuf at a given offset.
https://github.com/ipmitool/ipmitool/blob/master/lib/ipmi_fru.c#L656

If the offset is nonzero this will lead to a buffer overflow.

doublechiang, 2020-02-20:
Please correct me if I am wrong. 
It will check the fru size, and reduce the length to the maximum of fru size.
The code https://github.com/ipmitool/ipmitool/blob/master/lib/ipmi_fru.c#L822
```
	finish = offset + length;
	if (finish > fru->size) {
		memset(frubuf + fru->size, 0, length - fru->size);
		finish = fru->size;
		lprintf(LOG_NOTICE, "Read FRU Area length %d too large, "
			"Adjusting to %d",
			offset + length, finish - offset);
		length = finish - offset;
	}
```
But if the allocated memory size is less than the fru size, and the offset is not 0, then it might have some trouble. Any recommend approach to fix this issue?

CajuM, 2020-02-22:
would the following work? Before each read_fru_area
```
fru_data = malloc(fru_len + offset);
```
Also, shouldn't `length - fru->size` be `finish - fru->size` ?