ipmitool/ipmitool #146: IPMI tool core dump on host power off

gkeishin, 2019-07-11:
Started the IPMI SOL

```
ipmitool -I lanplus -C 3 -U root -P 0penBmc -H xx.xx.xx.xx sol activate
```
and was doing power on and off system and saw this one where it core dumped.

```
The system is going down for power-off NOW!

[   51.888284] systemd-shutdown[1]: Syncing filesystems and block devices.
[   52.011986] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[   52.048776] systemd-journald[4390]: Received SIGTERM from PID 1 (systemd-shutdow).
[   52.125908] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[   52.162332] systemd-shutdown[1]: Unmounting file systems.
[   52.162654] systemd-shutdown[19549]: Remounting '/' read-only in with options 'seclabel,attr2,inode64,noquota'.
[   52.464084] systemd-shutdown[1]: All filesystems unmounted.
[   52.464110] systemd-shutdown[1]: Deactivating swaps.
[   52.464164] systemd-shutdown[1]: All swaps deactivated.
[   52.464195] systemd-shutdown[1]: Detaching loop devices.
[   52.476238] systemd-shutdown[1]: All loop devices detached.
[   Segmentation fault (core dumped)
```
```
$ du -h core.17959
1.4M    core.17959
```
```
$ ipmitool -V
ipmitool version 1.8.18
```

- [ ] tar zipped file 
[core.17959.tar.gz](https://github.com/ipmitool/ipmitool/files/3382429/core.17959.tar.gz)

AlexanderAmelkin, 2019-07-11:
This is a known issue, a duplicate of https://sourceforge.net/p/ipmitool/bugs/495/
I believe you're using [OpenBMC](https://github.com/openbmc/openbmc/issues/1827), aren't you?

gkeishin, 2019-10-15:
yes @AlexanderAmelkin  on OpenBmc. any  potential fix coming there.