ipmitool/ipmitool #148: Windows binary with local connection support (FreeIPMI)

loop-evgeny, 2019-07-19:
Could we get a Windows binary (ipmitool.exe) published, please?

I managed to compile one under Cygwin, but it did not support managing IPMI on the local machine (only via network). I suppose this requires FreeIPMI, but I couldn't get this to compile under Windows, either. I'm sure someone has already solved this problem, as I see mentions of ipmitool.exe elsewhere, just no link to download it.

AlexanderAmelkin, 2019-09-06:
There is such thing as ipmitool by Supermicro. It has nothing to do with this project. Maybe you've seen references to it, although its full name is SMCIPMITool.exe. I'm not aware of any IPMI interface drivers for Windows except for Microsoft IPMI driver, which is not supported by this project.

Supermicro tools are suitable, AFAIK, for most IPMI-enabled systems, at least for the standardized features like setting IP addresses or SoL console.

closed 2020-07-03 by AlexanderAmelkin

loop-evgeny, 2020-07-06:
Thank you! One little typo: "**buildin** in Windows"

AlexanderAmelkin, 2020-07-06:
The typo is fixed now. Thank you. Does it work for you on Windows now? I believe you need to specify `--enable-intf-imb` to the `configure` script, and also have the Intel IMB driver installed, but I'm not sure because, as I said, I'm not familiar with Windows in general and with it on BMC-enabled servers in particular.

AlexanderAmelkin, 2020-07-06:
I don't see FreeIPMI library among the list of available cygwin packages, and I don't think that it's what you need.

loop-evgeny, 2020-07-06:
I've confirmed that it compiles, but as this was opened over a year I no longer remember which server we needed this on. We must have solved the problem using IPMICFG.exe from Supermicro. Thank you anyway!