ipmitool/ipmitool #150: Unable to establish IPMI connection to Dell R640

abhiniveshjain, 2019-08-04:
Getting below error while connecting to server. I have iDRAC 9.0 , IPMI version 2.0, I have also enabled IPMI over LAN setting.

ipmitool -I lanplus -H xx.xx.xx.xx -U xxxx -P xxxxx chassis status -v
Get Auth Capabilities error
Error issuing Get Channel Authentication Capabilities request
Error: Unable to establish IPMI v2 / RMCP+ session

AlexanderAmelkin, 2019-09-05:
Please run with `-vvvvvvvvv` (9 v's) and _attach_ (don't copy-paste) the log here.

talcoh2x, 2020-02-16:
coh_sw@HA01WVAW1993:~$ ipmitool -H 10.0.0.3 -U user -P useruser -I lanplus  power status -vvvvvvvvv

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x38
>>    data    : 0x8e 0x04

BUILDING A v1.5 COMMAND
added list entry seq=0x00 cmd=0x38
>> IPMI Request Session Header
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 00
>>   Rq Lun     : 0
>>   Command    : 38
>> sending packet (23 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 09 20 18
 c8 81 00 38 8e 04 b5
removed list entry seq=0x00 cmd=0x38

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x38
>>    data    : 0x8e 0x04

BUILDING A v1.5 COMMAND
added list entry seq=0x00 cmd=0x38
>> IPMI Request Session Header
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 00
>>   Rq Lun     : 0
>>   Command    : 38
>> sending packet (23 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 09 20 18
 c8 81 00 38 8e 04 b5
removed list entry seq=0x00 cmd=0x38

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x38
>>    data    : 0x8e 0x04

BUILDING A v1.5 COMMAND
added list entry seq=0x00 cmd=0x38
>> IPMI Request Session Header
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 00
>>   Rq Lun     : 0
>>   Command    : 38
>> sending packet (23 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 09 20 18
 c8 81 00 38 8e 04 b5
removed list entry seq=0x00 cmd=0x38

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x38
>>    data    : 0x8e 0x04

BUILDING A v1.5 COMMAND
added list entry seq=0x00 cmd=0x38
>> IPMI Request Session Header
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 00
>>   Rq Lun     : 0
>>   Command    : 38
>> sending packet (23 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 09 20 18
 c8 81 00 38 8e 04 b5
removed list entry seq=0x00 cmd=0x38

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x38
>>    data    : 0x0e 0x04

BUILDING A v1.5 COMMAND
added list entry seq=0x00 cmd=0x38
>> IPMI Request Session Header
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 00
>>   Rq Lun     : 0
>>   Command    : 38
>> sending packet (23 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 09 20 18
 c8 81 00 38 0e 04 35
removed list entry seq=0x00 cmd=0x38

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x38
>>    data    : 0x0e 0x04

BUILDING A v1.5 COMMAND
added list entry seq=0x00 cmd=0x38
>> IPMI Request Session Header
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 00
>>   Rq Lun     : 0
>>   Command    : 38
>> sending packet (23 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 09 20 18
 c8 81 00 38 0e 04 35
removed list entry seq=0x00 cmd=0x38

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x38
>>    data    : 0x0e 0x04

BUILDING A v1.5 COMMAND
added list entry seq=0x00 cmd=0x38
>> IPMI Request Session Header
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 00
>>   Rq Lun     : 0
>>   Command    : 38
>> sending packet (23 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 09 20 18
 c8 81 00 38 0e 04 35
removed list entry seq=0x00 cmd=0x38

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x38
>>    data    : 0x0e 0x04

BUILDING A v1.5 COMMAND
added list entry seq=0x00 cmd=0x38
>> IPMI Request Session Header
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 00
>>   Rq Lun     : 0
>>   Command    : 38
>> sending packet (23 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 09 20 18
 c8 81 00 38 0e 04 35
removed list entry seq=0x00 cmd=0x38
Get Auth Capabilities error
Error issuing Get Channel Authentication Capabilities request
Error: Unable to establish IPMI v2 / RMCP+ session
coh_sw@HA01WVAW1993:~$  sensor list   ^C

shenawy311, 2020-04-18:
Hello ,
I have the same issue. how your issue was solved ?

thanks

AlexanderAmelkin, 2020-07-16:
@shenawy311, are you using the github version or is your problem with the stock 1.8.18 in your OS repository?
Some IPMI server stacks (I can tell for OpenBMC for sure) have dropped support for some legacy insecure ciphers and the old versions of ipmitool couldn't properly detect the correct cipher to use. That was fixed in 7772254b62826b894ca629df8c597030a98f4f72 on Apr 9, 2018. Not all distributions may have picked up that patch. Please make sure you're using the latest sources from github.

AlexanderAmelkin, 2020-07-16:
First of all please make sure you're using the latest github code. Not the one from SourceForge, not the one you got from your OS vendor repository. Use github. The ouput pasted above (although I specifically asked to **attach** it) is not from the github version.

reesericci, 2020-07-26:
Is there any fix? I find it works on WSL, but not on bare-metal Ubuntu server

AlexanderAmelkin, 2020-07-28:
> Is there any fix? I find it works on WSL, but not on bare-metal Ubuntu server (using ubuntu repos)

I'm very sorry, but as I said earlier multiple times, please use the latest code from github, not from ubuntu repos.
If the problem is not reproducible with the latest code from this repository, but is reproducible with the binaries from ubuntu or whatever repos, then please ask for support from maintainers of those repos.

reesericci, 2020-08-02:
We figured out another way. Thx for the reply tho

closed 2020-08-12 by AlexanderAmelkin

ramanjk, 2021-06-18:
Hello @reesericci and @abhiniveshjain @AlexanderAmelkin , Can you please share the solution for this problem.

AlexanderAmelkin, 2021-06-18:
See #32