ipmitool/ipmitool #154: Unable to connect to supermicro, RAKP 2 fail

cerberek, 2019-09-10:
cannot log to Supermicro motherboard X10DRW-E.

With verbose output i got error:
RAKP 2 message indicates an error : inactive session ID
Error: Unable to establish IPMI v2 / RMCP+ session

Sometimes I got error: RAKP 2 HMAC is invalid
--------------------------------------------------------
ipmitool -vvvvv -C 3 -I lanplus -H xxx -U ADMIN -P xxx user list

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x38
>>    data    : 0x8e 0x04 

BUILDING A v1.5 COMMAND
added list entry seq=0x00 cmd=0x38
>> IPMI Request Session Header
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 00
>>   Rq Lun     : 0
>>   Command    : 38
>> sending packet (23 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 09 20 18
 c8 81 00 38 8e 04 b5
<< received packet (30 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 10 81 1c
 63 20 00 38 00 01 96 04 03 45 53 00 00 72
<< IPMI Response Session Header
<<   Authtype                : NONE
<<   Payload type            : IPMI (0)
<<   Session ID              : 0x00000000
<<   Sequence                : 0x00000000
<<   IPMI Msg/Payload Length : 16
<< IPMI Response Message Header
<<   Rq Addr    : 81
<<   NetFn      : 07
<<   Rq LUN     : 0
<<   Rs Addr    : 20
<<   Rq Seq     : 00
<<   Rs Lun     : 0
<<   Command    : 38
<<   Compl Code : 0x00
IPMI Request Match found
removed list entry seq=0x00 cmd=0x38
>> SENDING AN OPEN SESSION REQUEST

>> sending packet (48 bytes)
 06 00 ff 07 06 10 00 00 00 00 00 00 00 00 20 00
 00 00 00 00 a4 a3 a2 a0 00 00 00 08 01 00 00 00
 01 00 00 08 01 00 00 00 02 00 00 08 01 00 00 00
<< received packet (52 bytes)
 06 00 ff 07 06 11 00 00 00 00 00 00 00 00 24 00
 00 00 04 00 a4 a3 a2 a0 3f 8b 07 00 00 00 00 08
 01 00 00 00 01 00 00 08 01 00 00 00 02 00 00 08
 01 00 00 00
<<OPEN SESSION RESPONSE
<<  Message tag                        : 0x00
<<  RMCP+ status                       : no errors
<<  Maximum privilege level            : admin
<<  Console Session ID                 : 0xa0a2a3a4
<<  BMC Session ID                     : 0x00078b3f
<<  Negotiated authenticatin algorithm : hmac_sha1
<<  Negotiated integrity algorithm     : hmac_sha1_96
<<  Negotiated encryption algorithm    : aes_cbc_128

>> Console generated random number (16 bytes)
 b2 cc 52 38 70 d2 9d b5 34 08 a8 62 c2 e4 d2 1f
>> SENDING A RAKP 1 MESSAGE

>> sending packet (49 bytes)
 06 00 ff 07 06 12 00 00 00 00 00 00 00 00 21 00
 00 00 00 00 3f 8b 07 00 b2 cc 52 38 70 d2 9d b5
 34 08 a8 62 c2 e4 d2 1f 14 00 00 05 41 44 4d 49
 4e
<< received packet (24 bytes)
 06 00 ff 07 06 13 00 00 00 00 00 00 00 00 08 00
 00 08 00 00 00 00 00 00
<<RAKP 2 MESSAGE
<<  Message tag                   : 0x00
<<  RMCP+ status                  : inactive session ID
<<  Console Session ID            : 0x00000000
<<  BMC random number             : 0x008b0700000000080100000001000008
<<  BMC GUID                      : 0x01000000020000080100000000000000
<<  Key exchange auth code [sha1] : 0x0000000000000000000000000000000000000000

RAKP 2 message indicates an error : inactive session ID
Error: Unable to establish IPMI v2 / RMCP+ session
----------------------------------------------------------------------------------------

I tried to change password, factory default and flashing new firmware. Still same result. Any idea why I got this error?

gregorybrzeski, 2019-11-10:
Maybe an answer from Supermicro FAQ will help:

> This error is caused by a third party IPMI software that has RMCP+ (Remote Management Control Package +) built-in, which is not currently supported by our IPMI 2.0. Please use our IPMI View software to access the BMC card instead.

source: https://www.supermicro.com/support/faqs/faq.cfm?faq=3240

TomHetmer, 2020-06-22:
It seems like a supermicro firmware bug to me. Not even freeipmi has any luck with it

TomHetmer, 2020-06-22:
Actually it seems to happen when the ipmi is overloaded with requests as well (if you have a service like Foreman that runs too many ipmi commands at the host)

AlexanderAmelkin, 2020-08-03:
Closing it then.

closed 2020-08-03 by AlexanderAmelkin