ipmitool/ipmitool #158: Unable to change sensor threshold to negative value

RuudPendavingh, 2019-09-26:
The command to change a sensor threshold (e.g. of -12 V supply voltage) would be:
ipmitool -H <ip_addr> sensor thresh "Supply -12V" lcr -13.2
This command is not accepted as the new threshold value -13.2 is taken as an invalid option:
ipmitool: invalid option -- '1'
ipmitool version 1.8.18

AlexanderAmelkin, 2019-09-27:
This is a normal behavior for any Linux tool using getopt to parse the options. You can't start an argument with a dash/minus. You must tell getopt you're done with options. That is done by placing two dashes after the last option argument (or before the first non-option argument), like this:

```sh
ipmitool -H <ip_addr> -- sensor thresh "Supply -12V" lcr -13.2
```
Give it a try and tell me please if it helped.

RuudPendavingh, 2019-10-01:
This is really embarrassing: I already expected that ipmitool would use getopts and I did not check the manpage of getopts. In short: that works. Sorry to have wasted your time.

closed 2019-10-01 by RuudPendavingh