ipmitool/ipmitool #161: error: storage size of ‘ctx’ isn’t known

mihai-morman, 2019-10-15:
```
/bin/sh ../../../libtool --silent  --tag=CC   --mode=compile cc -DHAVE_CONFIG_H -I. -I../../..  -I../../../include   -g -O2 -Wall -Wextra -std=gnu99 -pedantic -Wformat -Wformat-nonliteral -MT lanplus_crypt_impl.lo -MD -MP -MF .deps/lanplus_crypt_impl.Tpo -c -o lanplus_crypt_impl.lo lanplus_crypt_impl.c
lanplus_crypt_impl.c: In function ‘lanplus_encrypt_aes_cbc_128’:
lanplus_crypt_impl.c:167:17: error: storage size of ‘ctx’ isn’t known
  167 |  EVP_CIPHER_CTX ctx;
      |                 ^~~
lanplus_crypt_impl.c:167:17: warning: unused variable ‘ctx’ [-Wunused-variable]
lanplus_crypt_impl.c: In function ‘lanplus_decrypt_aes_cbc_128’:
lanplus_crypt_impl.c:242:17: error: storage size of ‘ctx’ isn’t known
  242 |  EVP_CIPHER_CTX ctx;
      |                 ^~~
lanplus_crypt_impl.c:242:17: warning: unused variable ‘ctx’ [-Wunused-variable]
```
Hello!

I got that error after trying to install various versions of ipmitool on manjaro linux.
The above error is from _ipmitool-1.8.18.tar.gz_
I've tried also ipmitool versions 1.8.15, 1.8.17, with the same results.
What i did is:
`./configure`
`make`

My current kernel i 4.19.78-1-MANJARO.

I also want to point that ipmitool-1.8.18 works fine if i use pacman to install it.
I have some problems when uploading a firmware update using fwum 
I am getting Segmentation Fault:
`Segmentation fault (core dumped)                                          0 %`
`Segmentation fault (core dumped)######################################   99 %`

AlexanderAmelkin, 2019-10-23:
> What i did is:
> `./configure`
> `make`

Please read README and retry as described there.

AlexanderAmelkin, 2019-10-23:
Sorry, I meant `INSTALL`, not `README`.

HarpPDX, 2019-10-25:
I'm running into the same error. According to this (https://stackoverflow.com/questions/55992010/openssl-error-storage-size-of-ctx-isn-t-known), the issue is caused by a change in the openssl library.

I'm on Ubuntu 18.04, w/ gcc 7.4 and openssl 1.1.1.

I ran:
```
./bootstrap
./configure
make
```
I get:
```
<snip>
Making all in lanplus
make[4]: Entering directory '/home/me/ipmitool/src/plugins/lanplus'
/bin/bash ../../../libtool --silent  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../include   -g -O2 -Wall -Wextra -std=gnu99 -pedantic -Wformat -Wformat-nonliteral -MT lanplus_crypt_impl.lo -MD -MP -MF .deps/lanplus_crypt_impl.Tpo -c -o lanplus_crypt_impl.lo lanplus_crypt_impl.c
lanplus_crypt_impl.c: In function ‘lanplus_encrypt_aes_cbc_128’:
lanplus_crypt_impl.c:167:17: error: storage size of ‘ctx’ isn’t known
  EVP_CIPHER_CTX ctx;
                 ^~~
lanplus_crypt_impl.c:167:17: warning: unused variable ‘ctx’ [-Wunused-variable]
lanplus_crypt_impl.c: In function ‘lanplus_decrypt_aes_cbc_128’:
lanplus_crypt_impl.c:242:17: error: storage size of ‘ctx’ isn’t known
  EVP_CIPHER_CTX ctx;
                 ^~~
lanplus_crypt_impl.c:242:17: warning: unused variable ‘ctx’ [-Wunused-variable]
Makefile:441: recipe for target 'lanplus_crypt_impl.lo' failed
<snip>
```

junlizhang, 2020-02-22:
--- ipmitool-IPMITOOL_1_8_18-ori/src/plugins/lanplus/lanplus_crypt_impl.c       2016-10-08 16:17:55.000000000 +0800
+++ ipmitool-IPMITOOL_1_8_18-mod/src/plugins/lanplus/lanplus_crypt_impl.c       2020-02-22 17:35:09.526631700 +0800
@@ -164,16 +164,19 @@
                                                        uint8_t       * output,
                                                        uint32_t        * bytes_written)
 {
-       EVP_CIPHER_CTX ctx;
-       EVP_CIPHER_CTX_init(&ctx);
-       EVP_EncryptInit_ex(&ctx, EVP_aes_128_cbc(), NULL, key, iv);
-       EVP_CIPHER_CTX_set_padding(&ctx, 0);
+       EVP_CIPHER_CTX *ctx;
+       ctx = EVP_CIPHER_CTX_new();
+       EVP_CIPHER_CTX_init(ctx);
+       EVP_EncryptInit_ex(ctx, EVP_aes_128_cbc(), NULL, key, iv);
+       EVP_CIPHER_CTX_set_padding(ctx, 0);

        *bytes_written = 0;

-       if (input_length == 0)
+       if (input_length == 0){
+               EVP_CIPHER_CTX_free(ctx);
                return;
+       }

        if (verbose >= 5)
        {
@@ -191,26 +194,28 @@
        assert((input_length % IPMI_CRYPT_AES_CBC_128_BLOCK_SIZE) == 0);

-       if(!EVP_EncryptUpdate(&ctx, output, (int *)bytes_written, input, input_length))
+       if(!EVP_EncryptUpdate(ctx, output, (int *)bytes_written, input, input_length))
        {
                /* Error */
                *bytes_written = 0;
+               EVP_CIPHER_CTX_free(ctx);
                return;
        }
        else
        {
                uint32_t tmplen;

-               if(!EVP_EncryptFinal_ex(&ctx, output + *bytes_written, (int *)&tmplen))
+               if(!EVP_EncryptFinal_ex(ctx, output + *bytes_written, (int *)&tmplen))
                {
                        *bytes_written = 0;
+                       EVP_CIPHER_CTX_free(ctx);
                        return; /* Error */
                }
                else
                {
                        /* Success */
                        *bytes_written += tmplen;
-                       EVP_CIPHER_CTX_cleanup(&ctx);
+                       EVP_CIPHER_CTX_cleanup(ctx);
                }
        }
 }
@@ -239,10 +244,11 @@
                                                        uint8_t       * output,
                                                        uint32_t        * bytes_written)
 {
-       EVP_CIPHER_CTX ctx;
-       EVP_CIPHER_CTX_init(&ctx);
-       EVP_DecryptInit_ex(&ctx, EVP_aes_128_cbc(), NULL, key, iv);
-       EVP_CIPHER_CTX_set_padding(&ctx, 0);
+       EVP_CIPHER_CTX *ctx;
+       ctx = EVP_CIPHER_CTX_new();
+       EVP_CIPHER_CTX_init(ctx);
+       EVP_DecryptInit_ex(ctx, EVP_aes_128_cbc(), NULL, key, iv);
+       EVP_CIPHER_CTX_set_padding(ctx, 0);

        if (verbose >= 5)
@@ -255,8 +261,10 @@

        *bytes_written = 0;

-       if (input_length == 0)
+       if (input_length == 0){
+               EVP_CIPHER_CTX_free(ctx);
                return;
+       }

        /*
         * The default implementation adds a whole block of padding if the input
@@ -266,31 +274,33 @@
        assert((input_length % IPMI_CRYPT_AES_CBC_128_BLOCK_SIZE) == 0);

-       if (!EVP_DecryptUpdate(&ctx, output, (int *)bytes_written, input, input_length))
+       if (!EVP_DecryptUpdate(ctx, output, (int *)bytes_written, input, input_length))
        {
                /* Error */
                lprintf(LOG_DEBUG, "ERROR: decrypt update failed");
                *bytes_written = 0;
+               EVP_CIPHER_CTX_free(ctx);
                return;
        }
        else
        {
                uint32_t tmplen;

-               if (!EVP_DecryptFinal_ex(&ctx, output + *bytes_written, (int *)&tmplen))
+               if (!EVP_DecryptFinal_ex(ctx, output + *bytes_written, (int *)&tmplen))
                {
                        char buffer[1000];
                        ERR_error_string(ERR_get_error(), buffer);
                        lprintf(LOG_DEBUG, "the ERR error %s", buffer);
                        lprintf(LOG_DEBUG, "ERROR: decrypt final failed");
                        *bytes_written = 0;
+                       EVP_CIPHER_CTX_free(ctx);
                        return; /* Error */
                }
                else
                {
                        /* Success */
                        *bytes_written += tmplen;
-                       EVP_CIPHER_CTX_cleanup(&ctx);
+                       EVP_CIPHER_CTX_cleanup(ctx);
                }
        }

junlizhang, 2020-02-22:
Using EVP_CIPHER_CTX pionter implementation instead, IPMITOOL_1_8_18 and OpenSSL 1.1.1d verified.
[lanplus_crypt_impl.txt](https://github.com/ipmitool/ipmitool/files/4239068/lanplus_crypt_impl.txt)

HaoyuRuan, 2020-04-26:
`  //EVP_CIPHER_CTX ctx;
    EVP_CIPHER_CTX *ctx;
    ctx = EVP_CIPHER_CTX_new();`

elfosardo, 2020-05-04:
The fix is in master but unfortunately not in IPMITOOL_1_8_18, any chance to get a new release soon?

AlexanderAmelkin, 2020-05-25:
This has been fixed back in 2017 by commit 77fe5635. Closing.

closed 2020-05-25 by AlexanderAmelkin

hlfritz, 2021-09-21:
This may have resurfaced now in latest download today as I could not compile until I made the changes referenced above: 

https://github.com/ipmitool/ipmitool/commit/77fe5635037ebaf411cae46cf5045ca819b5c145