ipmitool/ipmitool #162: Any plans for a new release of ipmitool?

jvillal-amp, 2019-11-02:
The last release of ipmitool appears to be 1_8_18 which happened on 8-October-2016. Which is over three years ago.

Are there any plans to make a new release?

AlexanderAmelkin, 2019-11-06:
There are some issues that need to be fixed before a new release. But you're right, there's been a long time since Zdenek last released this project back on SourceForge. I really must think harder about making a new release.

tsdunc, 2020-02-06:
Agreed - It would be very helpful to have a release. How to convince OS vendors to pick up the latest changes? Ubuntu is still using 1.18.16 (not even the latest release).

jvillal-amp, 2020-02-07:
@AlexanderAmelkin I'll buy you lunch at Depo or Nikolsky Plaza if you make a release ;) I can actually do that since I am in Moscow at the moment.

AlexanderAmelkin, 2020-02-14:
My main concern is #177. Anyone willing to set up a virtual BMC and write automated tests?

Plus, there is a number of other issues I'd like to have resolved. At least #17 / #147. Unfortunately, I don't have time now to address those.

ajakk, 2020-07-26:
As annoying as it would be to produce a release that isn't as good as you think it could be, I really think it's worth putting off fixing those bugs for another release. Downstreams are currently shipping versions of ipmitool with [potential RCE vulnerabilities](https://github.com/ipmitool/ipmitool/security/advisories/GHSA-g659-9qxw-p7cp). I think such security issues would warrant a release, however imperfect, especially since it's been several months since the vulnerabilities were identified..

A couple downstream bugs for example:

https://bugs.gentoo.org/708436
https://security-tracker.debian.org/tracker/CVE-2020-5208 / https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=950761

doublechiang, 2021-05-22:
Vote for a new release. My opinion is a slight project doesn't do any good to the project too.

AlexanderAmelkin, 2022-08-30:
https://github.com/ipmitool/ipmitool/releases/tag/IPMITOOL_1_8_19

closed 2022-08-30 by AlexanderAmelkin