ipmitool/ipmitool #163: Unable to set options=efiboot,persistent

jvillal-amp, 2019-11-04:
I'm thinking the parsing of multiple "options=" values is somehow broken.

When I do the following:
```
ipmitool -I lanplus chassis bootdev disk options=persistent,efiboot -vvv
```

I see the following
```
<snip>
Chassis Set Boot Parameter 4 to 0101

>> Sending IPMI command payload
>>    netfn   : 0x00
>>    command : 0x08
>>    data    : 0x05 0xa0 0x08 0x00 0x00 0x00
<snip>
```
Notice the data is ``0x05 0xa0 ...``  0xA0 means EFI but NOT persistent.

When doing the command:
```
ipmitool -I lanplus chassis bootdev disk options=efiboot,persistent -vvv
```

I see:
```
<snip>
Chassis Set Boot Parameter 4 to 0101

>> Sending IPMI command payload
>>    netfn   : 0x00
>>    command : 0x08
>>    data    : 0x05 0xc0 0x08 0x00 0x00 0x00
<snip>
```
Notice the data is ``0x05 0xc0 ...``   0xC0 means persistent but NOT EFI

What we desire should look like this for either call:
```
Chassis Set Boot Parameter 4 to 0101

>> Sending IPMI command payload
>>    netfn   : 0x00
>>    command : 0x08
>>    data    : 0x05 0xe0 0x08 0x00 0x00 0x00
```
The data should be ``0x05 0xe0 ...``  0xE0 means EFI and persistent

The values are specified in Table 28-14 `Boot Option Parameters` in the IPMI Specification, V2.0, Rev. 1.1 https://www.intel.com/content/dam/www/public/us/en/documents/product-briefs/ipmi-second-gen-interface-spec-v2-rev1-1.pdf

AlexanderAmelkin, 2019-11-05:
Whoa! There is a whole nest of bugs in that parser and its counterpart!

The main bug that causes this particular issue is in line 1773 of ipmi_chassis.c where it says `flags[op->i] &= op->mask` instead of `flags[op->i] &= ~(op->mask)`, thus wiping out all other flags instead of clearing the state of the flag being set.

In addition to that, the code also confuses console redirection flags with boot verbosity flags and chokes on remote/redirected boot media flags.

I'm going to fix all that.

closed 2019-11-12 by AlexanderAmelkin

qthangnp, 2020-05-12:
When will the new ipmitool 1.8.19 released?

djuarezg, 2020-09-16:
I guess this has not yet been released, hasn't it?

doublechiang, 2020-09-16:
Not yet, you need to use the current master branch.

PR #165 (2019-11-12): Fix/cleanup chassis bootparam and bootdev options
This resolves #163 and also cleans up `chassis bootparam get` decoding and `chassis bootdev options=...` parsing.

Please read individual commit messages.