ipmitool/ipmitool #166: Is CVE-2011-4339 resolved?

JohnVillalovos, 2019-11-06:
Considering that there is some though about making a new release of ipmitool.

I see that Fedora has a patch applied against the 1_8_18 release to resolve CVE-2011-4339 

https://src.fedoraproject.org/rpms/ipmitool/blob/master/f/0001-CVE-2011-4339-OpenIPMI.patch

CVE Info here:
https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2011-4339

Is this already resolved or should that patch be applied?

As a note Fedora applies about 9 patches to the 1_8_18. Not sure if any of them should be incorporated into ipmitool or not.

Thanks!

AlexanderAmelkin, 2019-11-14:
As far as I can see, this issue has been fixed by Zdenek Styblik with ipmitool@0bbcc8f0cc34236d6f8549c5f51410420dbc8f08 back on Jan 24 2012 and does not exist in 1.8.18. The patch, however, may apply to 1.8.18 because the fix was in a different place.

AlexanderAmelkin, 2019-11-15:
Their patch 0002 is also not relevant (and will not apply).

jvillal-amp, 2019-11-15:
Thanks for the info. Sounds like things are good as far as CVEs then.

JohnVillalovos, 2019-11-15:
Thanks. Sounds like things are good for CVEs. Closing this issue.

closed 2019-11-15 by JohnVillalovos