ipmitool/ipmitool #167: Unable to edit "Board Mfg Date" field

mercier1200, 2019-11-06:
Hi,

it is more a question than a issue for now. I am unable to edit "Board Mfg Date" field using fru edit command. When typing ipmitool fru edit 0 field b 0 <something>, the field "Board Mfg" is modified. But according to the "IPMI Platform Management FRU Information Storage Definition v1.0 R1.1", you have 4 other fields before "Board Mfg". Since "Board Mfg" is listed as field #0, I obviously cannot pick a lower number to modify "Board Mfg Date" field.
[image]

Maybe there is another friendly method to edit this field? I know that I can use the raw command "write fru data" to specifically choose which bytes I want to change (which can be dangerous), I just wanted to ask if another method can be used.

Thanks is advance

AlexanderAmelkin, 2019-11-22:
Please see #172. It's requires changes to get merged, but you can build from that branch to get the functionality you need. Only works on x86 and other little-endian machines though.

mercier1200, 2019-11-27:
Ho nice, I didn't expect a PR for this, I will certainly use this functionality. Thanks!

bwil150n, 2020-03-30:
When can we look forward to seeing #172 released?