ipmitool/ipmitool #173: 'user set password' does only accept 15 users (mask 0x0F instead of 0x3F)

scotty-utb, 2019-11-26:
Hello,

using:
ipmitool -V
ipmitool version 1.8.18

ipmi-v2-rev1-1-spec-errata-6-markup.pdf
Table 22-34, Set User Password Command 
Byte 1 : User ID
[5:0] - User ID. 000000b = reserved. (User ID 1 is permanently associated 
with User 1, the null user name). 

ipmitool -I lanplus -C 1 -H <ip> -U <user> -P <pass> user set password 17 000000 -vvvvv
...
>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x47
>>    data    : **0x01** 0x02 0x30 0x30 0x30 0x30 0x30 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00

File:
lib/ipmi_user.c
Function:
_ipmi_set_user_password()
Line:
data[0] |= (0x0F & user_id);

User_id is limited to 4 byte, instead of allowed 6 byte.
Should be:
data[0] |= (0x3F & user_id);

scotty-utb, 2019-11-26:
Oh, this does duplicate this one:
https://github.com/ipmitool/ipmitool/issues/6

AlexanderAmelkin, 2019-11-27:
@scotty-utb, yep. Does it still happen for you with the latest master branch?

scotty-utb, 2019-11-27:
Yes, tested with the latest master branch now: solved.

closed 2019-11-27 by AlexanderAmelkin