ipmitool/ipmitool #177: Set up automated regression testing

AlexanderAmelkin, 2019-12-04:
Unfortunately `ipmitool` is not tested thoroughly enough. In fact only building is actually tested, and some functional tests are performed for the changed units manually during the PR merger process.

It is required to set up continuous integration regression testing for the project. Such testing would then be run for each pull request and separately before each release. The test framework must include a virtual BMC that would implement IPMI responses in strict accordance with the IPMI specification (ipmi_sim from OpenIPMI is one of the possibilities), and would also simulate OEM features. There must be a set of test cases developed and checked by the test.

This issue is to capture all the effort regarding implementation and setup of such a test framework.
Any comments and/or pull requests are welcome.

doublechiang, 2021-06-09:
I have a proposal to set up the regression test using ipmi_sim from OpenIPMI as BMC simulation.
Basically, it is build by:

1. git submodule to use the OpenIPMI project.
2. Build the ipmi_sim module and run it 
2. Then use RobotFramework to test the ipmitool do the regression test.
A working sample is available on my Github account https://github.com/doublechiang/ipmitool/tree/regression

Can you review to check if this works or anything needs to be changed before we start to develop on it?

AlexanderAmelkin, 2021-07-07:
The idea may be viable, I'm not very familiar with the OpenIPMI's ipmi_sim.
However, your example fails to build for me on Linux Mint 18.3 Sylvia (Ubuntu Bionic).

Also, I don't like the idea of storing openipmi as a submodule in ipmitool repo. Since it is only needed for the test, I think it can be fetched by the script. I would also prefer the script to be a part of the github CI workflow, see https://github.com/ipmitool/ipmitool/blob/master/.github/workflows/ci.yml
So it has to build for all the targets listed there or at least provide a server that all those builds could use to connect to for testing.

Please also note, that there is a very primitive simulator in the ipmitool project itself:
https://github.com/ipmitool/fake-ipmistack
However, in its current state it is definitely not suitable for any good testing as it only works over the 'dummy' protocol and covers a very limited subset of features.

doublechiang, 2022-03-30:
@AlexanderAmelkin I integrate the test into the Github Actions,  created another workflow file.
Would you check if this works? https://github.com/doublechiang/ipmitool/commits/regression

AlexanderAmelkin, 2022-03-30:
Thanks! Why don't you make a PR out of it for ease of review?