ipmitool/ipmitool #193: ipmitool with AD authentication

mskarthik28, 2020-04-10:
Hi Team, 

I am using ipmitool for setting ISO, change boot order and some other BIOS settings.currently we are using local ilo user(admin) to set the parameters through IPMITOOL. But now we will only support for AD authentication and going to remove local ilo admin user for some security reason. I am able to login through ilo GUI with my AD account and changing the BIOS settings, but i could not able to do through ipmitool with my AD account. i am getting following error while accessing through AD account. But if i am using local admin it is working fine through ipmitool.

Hardware : HP  ProLiant DL360/380 Gen10,Gen9,Gen8

Error message through AD account: 
=======================
BUILDING A v1.5 COMMAND
added list entry seq=0x00 cmd=0x38
>> IPMI Request Session Header
>>   Authtype   : NONE               ====> Here Authtype is using NONE with accessing AD account.
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000

Get Auth Capabilities error
Error issuing Get Channel Authentication Capabilities request
Error: Unable to establish IPMI v2 / RMCP+ session
=============================================
Working fine with local ilo admin user:
========================
<< IPMI Response Session Header
<<   Authtype                : RMCP+    =====>>  Auth type is using RMCP+
<<   Payload type            : IPMI (0)
<<   Session ID              : 0xa0a2a3a4
<<   Sequence                : 0x00000007
<<   IPMI Msg/Payload Length : 32

ipmitool -H 192.168.0.10 -U xzy123 -P Password -I lanplus power status

Example: 
xzy123 - AD account
password - Password
192.168.0.10 - Remote console IP
=============================================

Kindly help me how to use AD account through IPMITOOL.

doublechiang, 2020-04-10:
I believed the authentication is processed in the BMC itself. On the Quanta systems I have, there is no problem to be authenticated by AD. IPMITOOL had send the correct RMCP protocol. Have you checked any updated BMC firmware on your HP systems?

mskarthik28, 2020-04-10:
Yes, i have checked the version , it is Firmware Revision: 2.70. YIPMITOOL use RMPC protocol but it works with local account , if i use AD account it says NONE protocol type. Is there anything needs to be change in ILO settings. Hardware model :Product Name: ProLiant DL360 Gen9 , Ilo4.

AlexanderAmelkin, 2020-04-10:
ipmitool doesn't do any magic, it just sends commands prescribed by the IPMI protocol specification. All authentication is done according to that specification.

I believe AD can not be used for IPMI-enabled accounts because the IPMI protocol specification implies that the BMC has a copy of plain-text password stored, which is not possible with any LDAP-based authentication.

Try using RedFish or other protocols that your HP iLO may be providing.

Bottom line: nothing can be done on ipmitool side about it.

closed 2020-04-10 by AlexanderAmelkin

mskarthik28, 2020-04-10:
Thanks for the information Alex. So it will support all authentication through IPMITOOL. Is there anyway to achieve it  apart from RedFish. ? If we use plain text password , will it work.

AlexanderAmelkin, 2020-04-14:
Please address your questions to HP. I unfortunately have no idea how to answer them.
If your iLO has an IPMI user, ipmitool will be able to authenticate. It's that simple. Any questions on how to set up anything in iLO should be addressed to iLO guys. Sorry.