ipmitool/ipmitool #195: Error "Invalid argument" when run IPMI command "Ipmitool raw 0x06 0x35"

ThuBaNguyen, 2020-04-15:
On Ubuntu 18.04 and CentOS 7, 8. 
IPMITool version 1.8.18.

When I run IPMI command "Ipmitool raw 0x06 0x35" I got below error:
Unable to send command: Invalid argument
Unable to send RAW command (channel=0x0 netfn=0x6 lun=0x0 cmd=0x35)

With debug option
#ipmitool raw 0x06 0x35 -vvv
Using ipmi device 0
Set IPMB address to 0x20
OpenIPMI Request Message Header:
  netfn     = 0x6
  cmd       = 0x1
Iana: 11183
Running Get PICMG Properties my_addr 0x20, transit 0, target 0
OpenIPMI Request Message Header:
  netfn     = 0x2c
  cmd       = 0x0
OpenIPMI Request Message Data (1 bytes)
 00
Error response 0xc1 from Get PICMG Properities
Running Get VSO Capabilities my_addr 0x20, transit 0, target 0
OpenIPMI Request Message Header:
  netfn     = 0x2c
  cmd       = 0x0
OpenIPMI Request Message Data (1 bytes)
 03
Invalid completion code received: Invalid command
Acquire IPMB address
Discovered IPMB address 0x0
Interface address: my_addr 0x20 transit 0:0 target 0x20:0 ipmb_target 0

RAW REQ (channel=0x0 netfn=0x6 lun=0x0 cmd=0x35 data_len=0)
OpenIPMI Request Message Header:
  netfn     = 0x6
  cmd       = 0x35
Unable to send command: Invalid argument
Unable to send RAW command (channel=0x0 netfn=0x6 lun=0x0 cmd=0x35)

IPMITool server don't get request with netfn =0x6 and cmd=0x35 from host.

doublechiang, 2020-04-16:
BMC got this command request, but it doesn't implement this handler for this command. Ipmitool does nothing related to this issue.

ThuBaNguyen, 2020-04-17:
Add debug code in BMC side.
Test case 1: Run IPMI command with lanplus interface: I got error
# ipmitool <BMCIP> raw 0x6 0x31
Unable to send RAW command (channel=0x0 netfn=0x6 lun=0x0 cmd=0x31 rsp=0xd4): Insufficient privilege level
--> BMC handle this command and report error because this "Insufficient privilege level".
Test case 2: Run IPMI command with system interface: I got error:
Unable to send command: Invalid argument
Unable to send RAW command (channel=0x0 netfn=0x6 lun=0x0 cmd=0x35)
--> BMC Message handler is not called.

AlexanderAmelkin, 2020-05-25:
You're right, the `Unable to send command: Invalid argument` message may mean that the command has not been sent to the BMC. Actually, it mean that the system call to the OpenIPMI device node (`/dev/ipmi0` usually) has failed with an `EINVAL` return code. Most probably there is something wrong with your IPMI device.

I've did a quick check on the Linux kernel drivers/char/ipmi/ipmi_ssif.c and I see that it reports EINVAL when it can't parse the response from the BMC. The lanplus and host requests may be processed differently by your BMC, so maybe it gives a bad response even before reaching your debug statements in the handler.

Or maybe, you just don't have a correct `/dev/ipmi0` device served by the OpenIPMI driver and whatever is serving that node just wants to say that `ipmitool`'s request to send a command wasn't recognized.

Anyway, this doesn't look like `ipmitool`'s problem so far. Sending RAW commands via OpenIPMI interface have been working for ages now.

ThuBaNguyen, 2020-05-26:
Thank for your response Alexander.
Maybe you was right, I though in ipmitool side, there are no difference when it parse command "raw 0x06 0x31" and "raw 0x06 0x35".
If it worked correctly with "raw 0x06 0x31" then It should work with "raw 0x06 0x35".
Let me close the issue and debug more in code section before ipmi handler.

closed 2020-05-26 by ThuBaNguyen