ipmitool/ipmitool #196: IPMItool has not support Cipher Suite 11, 12

hchieu, 2020-04-16:
Hi IPMItool supporter,

Currently, our BMC support Cipher Suites: 1,2,3,6,7,8,11,12,15,16,17
```
$ ipmitool lan print
RMCP+ Cipher Suites     : 1,2,3,6,7,8,11,12,15,16,17
Cipher Suite Priv Max   : XXaXXXXXXXaXXXX
                        :     X=Cipher Suite Unused
                        :     c=CALLBACK
                        :     u=USER
                        :     o=OPERATOR
                        :     a=ADMIN
                        :     O=OEM

```
When we enable another Cipher suite e.g1,2,6,7,8,11,12,15,16 by command:
```
$ Ipmitool -C 3 lan set 1 cipher_privs aaaaaaaaaaaXXXX
$ ipmitool lan print
RMCP+ Cipher Suites     : 1,2,3,6,7,8,11,12,15,16,17
Cipher Suite Priv Max   : aaaaaaaaaaaXXXX
                        :     X=Cipher Suite Unused
                        :     c=CALLBACK
                        :     u=USER
                        :     o=OPERATOR
                        :     a=ADMIN
                        :     O=OEM
```
Verify Cipher Suite by command:
`$ ipmpitool -C 11 -L ADMIN chassis power status`
The problem is: we can verify the CS1,2,3,6,7,8,15,16,17 except CS11, CS12.
```
ipmpitool -C 11 -L ADMIN chassis power status
ipmitool: lanplus.c:1907: ipmi_lanplus_build_v2x_msg: Assertion `0' failed.
Aborted (core dumped)
```
I find the code in:
```
switch(session->v2_data.integrity_alg) {
 case IPMI_INTEGRITY_HMAC_SHA1_96:
 assert(hmac_length == IPMI_SHA_DIGEST_LENGTH);
 auth_length = IPMI_SHA1_AUTHCODE_SIZE;
 break;
 case IPMI_INTEGRITY_HMAC_MD5_128 :
 assert(hmac_length == IPMI_MD5_DIGEST_LENGTH);
 auth_length = IPMI_HMAC_MD5_AUTHCODE_SIZE;
 break;
#ifdef HAVE_CRYPTO_SHA256
 case IPMI_INTEGRITY_HMAC_SHA256_128:
 assert(hmac_length == IPMI_SHA256_DIGEST_LENGTH);
 auth_length = IPMI_HMAC_SHA256_AUTHCODE_SIZE;
 break;
#endif /* HAVE_CRYPTO_SHA256 */
 default:
 assert(0);
 break;
 }
```
The Cipher Suite ID 11, 12 have Integrity Algorithms: MD5-128.
There is no processed for case MD5-128.

And [ID:441:] (https://github.com/ipmitool/ipmitool/commit/8ca47f21ca9e121ceabf3eab68244206c5a5b7ae) But has not support for CS11, CS12.

Why you do not support CS11, CS12?. And how can we support it?

vmauery, 2020-04-21:
Basically all of the cipher suites except 3 and 17 are really quite insecure and should not ever be used. Cipher suite 3 is suspect and on the weak side, so 17 should be used whenever possible. I don't have any plans to add support for a cipher suite that uses insecure mechanisms for authentication and integrity.

AlexanderAmelkin, 2020-05-22:
Vernon's comment explains it all. No plans to add support for obsolete and insecure cipher suites that won't be used.

closed 2020-05-22 by AlexanderAmelkin