ipmitool/ipmitool #203: Issue: lanplus feature in ipmitool v 1.8.18 haven't been fixed (Window 10 env)

nhanlee97, 2020-05-24:
Hello everyone,

I'm building the ipmitool version 1.8.18 for window environment (env). Build complete ! (Build on Cygwin application). When I run ipmitool -H xxx -U yyy -P zzz -I **lanplus** sth  (not work -- console shows segmentation (core dump)). But I run this command and replace **lanplus** to **lan**, then it works. So please help me if you can? Note: I want build ipmitool for Window env

Thanks in advance,
nhanlee97

nhanlee97, 2020-05-24:
Hi every one,

I just tried to edit by reinstall with other openssl-devel library (version 1.1.1d). Build completely? But I have not still used **lanplus** feature. 

Thansks in advance,
nhanlee97

AlexanderAmelkin, 2020-05-25:
Unfortunatly, Windows support is not maintained as I don't have Windows, and there are no official instructions on how to build for Windows. What is the procedure that you're using?
I will then try to check it on some VM and then maybe add an automated build test procedure to avoid breaking the Windows build with future commits.

HaoyuRuan, 2020-05-26:
using cygwin
1、install build tool and lib
C:\cygwin>setup-x86.exe -q -P wget -P gcc-g++ -P make -P diffutils -P libmpfr-devel -P libgmp-devel -P libmpc-devel -P openssl-devel

2、configure enable lanplus
./configure --enable-intf-lanplus
3、make 
  make

nhanlee97, 2020-05-26:
Hello everyone,
I will test right now. If I use completely lanplus feature, I will close this issue. Thanks for your support!
Best regards,
nhanlee97

AlexanderAmelkin, 2020-05-26:
> 2、configure enable lanplus
> ./configure --enable-intf-lanplus

Before configure is even available, you need to run `./bootstrap`, and that fails for me on Windows.
Can you help me fix the Travis-CI script for Windows?
The failed job can be found here:
https://travis-ci.com/github/ipmitool/ipmitool/jobs/339998608

nhanlee97, 2020-05-27:
Hi AlexanderAmelkin,

I downloaded ipmitool for window at link (https://sourceforge.net/projects/ipmitool/).

Best regards,
nhanlee97

nhanlee97, 2020-05-27:
> using cygwin
> 1、install build tool and lib
> C:\cygwin>setup-x86.exe -q -P wget -P gcc-g++ -P make -P diffutils -P libmpfr-devel -P libgmp-devel -P libmpc-devel -P openssl-devel
> 
> 2、configure enable lanplus
> ./configure --enable-intf-lanplus
> 3、make
> make

Hi HaoyuRuan,

I have still met this issue. I attached pictures below for your review. After config follow your guide, build completely, and **lanplus** feature have not still worked. 😞 

Best regards,
nhanlee97

doublechiang, 2020-06-04:
> Before configure is even available, you need to run `./bootstrap`, and that fails for me on Windows.
> Can you help me fix the Travis-CI script for Windows?
> The failed job can be found here: [...]

To be able to run the ./bootstrp. download more package in the script.
However, in my environment, "configure will not success unless specify "--disable-dependency-tracking"
```
setup-x86_64.exe -q -P automake -P autoconf -P libtool -P libreadline-devel
```

PuwenHsiao, 2020-06-05:
Hi ,
This is my note for building IPMITOOL under cygwin, but it was written by Chinese.
https://ipmitool.blogspot.com/2016/04/windows-os-ipmitool.html?m=1
 
You may reference the installation steps below:

*** Download cygwin ***
https://www.cygwin.com/setup-x86_64.exe

*** Install cygwin ***
$ setup-x86_64 --quiet-mode --disable-buggy-antivirus --upgrade-also --wait --arch="x86" --root="c:\cygwin" --only-site --site="ftp://ftp.cse.yzu.edu.tw/pub/cygwin/" --packages="bash,p7zip,libssl-devel,git,gcc-core,make,diff,autoconf,automake,m4,libtool,libncurses-devel,libreadline-devel" --verbose --no-admin

*** !!! Clone project in cygwin Terminal ***

$ git clone https://github.com/ipmitool/ipmitool.git
$ cd ipmitool

*** Build the IPMITOOL ***
$ ./bootstrap
$ ./configure IANADIR="./" IANAUSERDIR="./" DEFAULT_INTF="lanplus" --enable-intf-lan --enable-intf-lanplus --disable-intf-bmc --disable-intf-free --disable-intf-open --disable-intf-imb --disable-intf-usb --enable-intf-serial --enable-ipmishell
$ LANG=C make

AlexanderAmelkin, 2020-06-05:
> Hi AlexanderAmelkin,
> 
> I downloaded ipmitool for window at link (https://sourceforge.net/projects/ipmitool/).

There is no `ipmitool for windows` there, there is just the sources, and quite outdates sources I must say. You better download the latest souces from here:
https://github.com/ipmitool/ipmitool/archive/master.zip

Anyway, I don't think even Zdenek, when he maintained the project, verified it for full Windows compatibility. Neither do I. As I said, there is a branch with Travis CI configuration or building for Windows, but it doesn't build right now. Anyone willing to fix it is welcome to create a PR against this branch:
https://github.com/ipmitool/ipmitool/tree/feature/travis-cygwin

AlexanderAmelkin, 2020-07-03:
Building with Travis CI for Windows turned out to be a real pain in the butt.
Thankfully, there is now GitHub Actions / Workflows, and CI for Windows has been setup with them and verified.
Please refer to the instructions given in `INSTALL` file on how to build for Windows.

closed 2020-07-03 by AlexanderAmelkin