ipmitool/ipmitool #205: can't build on alpine

mmlb, 2020-05-27:
In https://github.com/ipmitool/ipmitool/commit/95038ba01b99153de870dde91406bdc1c7265217#diff-67e997bcfdac55191033d57a16d1408aR690 a bashism was introduced into configure.ac that causes ipmitool to not build in alpine (at least anything > 3.8) and likely debian too (due to sh -> dash).

I'm thinking about to how to get configure to work in alpine's ash based sh, but wanted to open this up for discussion/brainstorming.

AlexanderAmelkin, 2020-06-04:
Please check if that branch works for you

Chocobo1, 2020-06-14:
Hi, I'm building on archlinux and I believe I'm having the same problem.
This is the last few lines after issuing `./bootstrap &&  ./configure --prefix="/usr"`:
```
checking for OpenIPMI dual bridge support... no
checking sys/lipmi/lipmi_intf.h usability... no
checking sys/lipmi/lipmi_intf.h presence... no
checking for sys/lipmi/lipmi_intf.h... no
** Unable to build Solaris 9 x86 IPMI interface support!
checking for library containing readline... -lreadline
./configure: 14415: test: x: unexpected operator
configure: error: ** Cannot set  as default; intf- is not enabled.
==> ERROR: A failure occurred in build().
    Aborting...
```
`configure` file at line 14415:
```
if test "x${DEFAULT_INTF}" == "x"; then
	echo "DEFAULT_INTF not found in environment; setting to ${DEFAULT_INTF_NO_ENV}"
	DEFAULT_INTF=${DEFAULT_INTF_NO_ENV}
fi
```
Also note that the patch in #207 is included.

It seems to me that `==` should be `=` instead.

AlexanderAmelkin, 2020-06-15:
> It seems to me that `==` should be `=` instead.

Yep, looks like you're right. My bad.

AlexanderAmelkin, 2020-06-15:
There was also another problem with #207 that is now fixed there. Please check.

Chocobo1, 2020-06-16:
>There was also another problem with #207 that is now fixed there. Please check.

👍
Tested, now I can build ipmitool successfully.

closed 2020-06-16 by AlexanderAmelkin