ipmitool/ipmitool #206: Tiny error in Makefile

patch-work, 2020-05-29:
```
> git clone https://github.com/ipmitool/ipmitool.git
> ./bootstrap
```

...

Makefile:
```
install-pen-database: enterprise-numbers
        mkdir -m 755 -p $(DESTDIR)$(IANADIR)
        $(INSTALL_DATA) $< -t $(DESTDIR)$(IANADIR)/
...........................^^^
```
...

`> sudo make install`
...
```
mkdir -m 755 -p /usr/share/ipmitool/misc
/usr/bin/install -c -m 644 enterprise-numbers -t /usr/share/ipmitool/misc/
install: -t: No such file or directory
make[2]: *** [install-pen-database] Error 71
```
`> man install`

```
SYNOPSIS
     install [-bCcMpSsv] [-B suffix] [-f flags] [-g group] [-m mode] [-o owner] file1 file2
     install [-bCcMpSsv] [-B suffix] [-f flags] [-g group] [-m mode] [-o owner] file1 ... fileN directory
     install -d [-v] [-g group] [-m mode] [-o owner] directory ...
```

doublechiang, 2020-06-02:
May I know what parameter you used in the "configure"?
I have no problem to running the make install command on Debian 9.
```
mkdir -m 755 -p /usr/local/share/misc
/usr/bin/install -c -m 644 enterprise-numbers -t /usr/local/share/misc/
```

AlexanderAmelkin, 2020-06-04:
> `> man install`
> 
> ```
> SYNOPSIS
>      install [-bCcMpSsv] [-B suffix] [-f flags] [-g group] [-m mode] [-o owner] file1 file2
>      install [-bCcMpSsv] [-B suffix] [-f flags] [-g group] [-m mode] [-o owner] file1 ... fileN directory
>      install -d [-v] [-g group] [-m mode] [-o owner] directory ...
> ```

```
$ apt contains `which install`
coreutils: /usr/bin/install
$ apt list coreutils
Listing... Done
coreutils/bionic,now 8.28-1ubuntu1 amd64 [installed]
$ man install
INSTALL(1)                    User commands                     INSTALL(1)

NAME
       install - copy files and set attributes

SYNOPSIS
       install [OPTION]... [-T] SOURCE DEST
       install [OPTION]... SOURCE... DIRECTORY
       install [OPTION]... -t DIRECTORY SOURCE...
       install [OPTION]... -d DIRECTORY...
```
I also checked other flavors of Linux and they all had `install` as part of `coreutils` and with the `-t` option for the target directory. Removing it will break all Linux systems.

What system are you using where `install` doesn't have the `-t` option? Is it some kind of BSD or what? macOS build and installation is verified by TravisCI and passes without problems.

patch-work, 2020-06-04:
OpenBSD

AlexanderAmelkin, 2020-06-30:
It turns out the `-t` is also not supported by `install` on macOS (Xcode 10.15).
I've looked at the Linux man page more closely and found out that the `-t` option is quite, well, optional. :)
I now think that removing it may be safe, and I'll do it.

closed 2020-07-03 by AlexanderAmelkin