ipmitool/ipmitool #210: Error setting threshold: Command illegal for specified sensor or record type while changing CPU sensor threshold

asaki1986, 2020-06-29:
Hi,

I am using ipmitool to change the threshold for the 'Temp' sensor.

Commands and  returned messages are listed as follows,

sudo ipmitool sensor list | grep Temp
`Inlet Temp       | 16.000     | degrees C  | ok    | na        | -7.000    | 3.000     | 43.000    | 47.000    | na`
`Exhaust Temp     | 32.000     | degrees C  | ok    | na        | 3.000     | 8.000     | 70.000    | 75.000    | na`
`Temp             | 41.000     | degrees C  | ok    | na        | 3.000     | 8.000     | 79.000    | 84.000    | na`
`Temp             | 35.000     | degrees C  | ok    | na        | 3.000     | 8.000     | 79.000    | 84.000    | na`

`sudo ipmitool sensor thresh 'Temp' upper 85 92 90`
`Locating sensor record 'Temp'...`
`Setting sensor "Temp" Upper Non-Critical threshold to 85.000`
`Error setting threshold: Command illegal for specified sensor or record type`
`Setting sensor "Temp" Upper Critical threshold to 92.000`
`Error setting threshold: Command illegal for specified sensor or record type`
`Setting sensor "Temp" Upper Non-Recoverable threshold to 90.000`
`Error setting threshold: Command illegal for specified sensor or record type`

Any suggestions for the right commands?

Thanks.

doublechiang, 2020-06-29:
There is command line help you can refer to.

$ ipmitool sensor thresh
sensor thresh <id> <threshold> <setting>
   id        : name of the sensor for which threshold is to be set
   threshold : which threshold to set
                 unr = upper non-recoverable
                 ucr = upper critical
                 unc = upper non-critical
                 lnc = lower non-critical
                 lcr = lower critical
                 lnr = lower non-recoverable
   setting   : the value to set the threshold to

sensor thresh <id> lower <lnr> <lcr> <lnc>
   Set all lower thresholds at the same time

sensor thresh <id> upper <unc> <ucr> <unr>
   Set all upper thresholds at the same time

asaki1986, 2020-06-29:
> There is command line help you can refer to.
> 
> $ ipmitool sensor thresh
> sensor thresh
> id : name of the sensor for which threshold is to be set
> threshold : which threshold to set
> unr = upper non-recoverable
> ucr = upper critical
> unc = upper non-critical
> lnc = lower non-critical
> lcr = lower critical
> lnr = lower non-recoverable
> setting : the value to set the threshold to
> 
> sensor thresh lower
> Set all lower thresholds at the same time
> 
> sensor thresh upper
> Set all upper thresholds at the same time

I actually run the same command, "sudo ipmitool sensor thresh 'Temp' upper 85 92 90", but get "Error setting threshold: Command illegal for specified sensor or record type" as feedback.

doublechiang, 2020-06-29:
your command should be looks like to "ipmitool sensor thresh Temp ucr 90"

asaki1986, 2020-06-29:
> your command should be looks like to "ipmitool sensor thresh Temp ucr 90"

get the same message using the command you provide

`sudo ipmitool sensor thresh Temp ucr 90`
Locating sensor record 'Temp'...
Setting sensor "Temp" Upper Critical threshold to 90.000
Error setting threshold: Command illegal for specified sensor or record type

doublechiang, 2020-06-29:
I apologized, your command is right the first time. I missed the part you can set the threshold at the same time. 
I will guess your system (BMC) does not support 'Set Sensor Threshold' (0x26) command.

asaki1986, 2020-06-30:
> I apologized, your command is right the first time. I missed the part you can set the threshold at the same time.
> I will guess your system (BMC) does not support 'Set Sensor Threshold' (0x26) command.

Any method to allow or add this support?

doublechiang, 2020-06-30:
I will assume it is related to your system's vendor issue. This is not related to ipmitool, since ipmitool works well on my system.

AlexanderAmelkin, 2020-07-03:
@asaki1986, as properly noted by @doublechiang, you should direct this question to your BMC vendor.
I would expect that changing sensor threshold was not supported because supporting this feature would pose a hardware vitality risk, and any sane vendor would prohibit changing the thresholds, at least the non-recoverable and the critical ones.

Try setting just the upper non-critical threshold like this:
`sudo ipmitool sensor thresh 'Temp' unc 85`

In any case, please ask your vendor for further assistance.

closed 2020-07-03 by AlexanderAmelkin