ipmitool/ipmitool #213: [ipmitool] question about internals on closing IPMI sessions

zerodayz, 2020-07-06:
Hello,

This is rather a question than an issue. I would like to confirm the behaviour of ipmitool when we encounter following:

- Slow BMC responses
- Dropped packets

While ipmitool allows usage of "-R" and "-N" respectively, what happens if the connection reaches out time-out value of "-N" due to one of the reasons specified above.

Does it open another session towards the BMC or close down the session first and opens a new one?

The reason why I am asking is that, some BMC consoles *Cisco IMC* doesn't allow to have more than 4 sessions open.

I am absolutely sure that someone already hit this problem.

Thank you for your reply.

zerodayz, 2020-07-06:
/label question

zerodayz, 2020-07-09:
@AlexanderAmelkin 

We can see that the connection to BMC is opened, then BMC even correctly responds with the status. Then tries to close the session and never gets a response back from BMC.

Since we are using UDP, which is fast but not reliable. It may be the packets are simply not delivered due to Network issues. Then session cannot me closed properly which leads to max sessions open in Cisco IMC. This kills all sessions that are inactive for 60s. 

Is there any possible improvement ?

~~~
[ 2020-07-09 04:10:20,030 DEBUG: 0 Chassis Power is on ]
[  Close Session command failed ]
~~~

https://github.com/ipmitool/ipmitool/blob/2ee7aca4e4c40490459387d82d87bb4cd2f1a898/src/plugins/lanplus/lanplus.c#L2739-L2766

From the ipmitool code, it's part of the ipmi_close_session_cmd function.

~~~
static int
ipmi_close_session_cmd(struct ipmi_intf * intf)
{
...
	rsp = intf->sendrecv(intf, &req);
	if (!rsp) {
		/* Looks like the session was closed */
		lprintf(LOG_ERR, "Close Session command failed");
		return -1;
	}
~~~

steveb, 2020-10-02:
The -R and -N options are used in the sendrecv function, which ipmi_close_session_cmd calls. So the only real option for losing close request packets or slow BMCs is to increase -R.