ipmitool/ipmitool #218: Replace curl argument

Ypnose, 2020-07-20:
Hey,
When attempting to download IANA PEN database with `curl`, it might fail with this error message :
```
Downloading IANA PEN database...
curl: option -: is unknown
curl: try 'curl --help' or 'curl --manual' for more information
FAILED to download the IANA PEN database
```
It is probably related to `-#` argument in `curl` command, which is launched without escaping. Would you agree to replace it by `--progress-bar` ? It's the human friendly counterpart option. I can submit a pull request if you like.

AlexanderAmelkin, 2020-07-20:
@Ypnose Seems reasonable, please submit a PR

Ypnose, 2020-07-21:
Please wait before applying pull request, I just noticed the link has changed. `curl` command needs to be updated with `-L` too.
```
$ curl -LI http://www.iana.org/assignments/enterprise-numbers
HTTP/1.1 301 Moved Permanently
Date: Tue, 21 Jul 2020 05:30:35 GMT
Server: Apache
X-Frame-Options: SAMEORIGIN
Referrer-Policy: origin-when-cross-origin
Content-Security-Policy: upgrade-insecure-requests
Location: https://www.iana.org/assignments/enterprise-numbers/enterprise-numbers
Cache-Control: public, s-maxage=1800, max-age=3600
Expires: Tue, 21 Jul 2020 06:30:35 GMT
Content-Type: text/html; charset=iso-8859-1
```

AlexanderAmelkin, 2020-07-21:
Sure, do it in the same PR please.

AlexanderAmelkin, 2020-07-21:
Also, to block a PR, you may just add "WIP" to it's title. That will fail the WIP plugin check and prevent merging.

Ypnose, 2020-07-21:
Thanks for the insight. I submitted this change in the same PR. As the checks are all good, I took the initiative to remove "WIP" from the title. It sounds all good.

PR #219 (2020-07-21): configure.ac: replace '-#' by '--progress-bar' with curl
It fixes #218
closed 2020-07-21 by AlexanderAmelkin