ipmitool/ipmitool #220: fail to build from source with GCC-10

JoergFF, 2020-07-28:
Hi, 
In Debian Sid we get a bug in the compiling with gcc-10. 
The build log is attached.

The Debian Bug Report is https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=957371
CU
Jörg

[ipmitool.build.log](https://github.com/ipmitool/ipmitool/files/4987339/ipmitool.build.log)

AlexanderAmelkin, 2020-07-28:
This:
```
/usr/bin/ld: ../lib/.libs/libipmitool.a(ipmi_hpmfwupg.o):./lib/../include/ipmitool/ipmi_hpmfwupg.h:803: multiple definition of `gVersionInfo'; ipmitool.o:./src/../include/ipmitool/ipmi_hpmfwupg.h:803: first defined here
```
looks a lot like a bug in gcc-10 because it's not "multiple definitions", but the same file.

Nonetheless, that declaration is clearly not needed in the header, as well as the `errorCount` variable.

Please check this branch and report if it helps (I don't have gcc10):
https://github.com/ipmitool/ipmitool/tree/bugfix/220-fix-gcc10

AlexanderAmelkin, 2020-07-28:
By the way, you're free to submit a PR with a github workflow addition to automatically check building with gcc10 on Ubuntu or even Debian Sid (the latter requires a hosted VM though).

JoergFF, 2020-07-28:
Hi Alexander,

here are my patch for the gcc-10 bug:

CU
Jörg

[0005-gcc10.zip](https://github.com/ipmitool/ipmitool/files/4989743/0005-gcc10.zip)

AlexanderAmelkin, 2020-07-28:
> 0005-gcc10.zip

Doesn't my fix from the above branch work for you?

AlexanderAmelkin, 2020-07-28:
I mean commits 0e128b3c9bdbbf9a6c47be3209b99b081c9aad96 and 702080970e3f5a9d56ba62baba48279724abf55f

JoergFF, 2020-07-28:
Hi Alexander,

your commits not working. I think you must only one without and all other includes with 'extern'.
But I don't check it with all 'extern' . I do it tomorrow.

CU
Jörg

AlexanderAmelkin, 2020-07-31:
This is very strange. First of all, I can not reproduce your problem with the current master branch and gcc-10 on my Linux Mint 19.1. Second, my changes in that branch completely removed what your changes are trying to conditionally comment out, thus if your changes work, mine must work too.

I still suspect it is some bug in your environment because from the C language point of view there were no bugs in that part of code.

yontalcar, 2020-08-07:
I already fixed the extern issue in master with 51c7e0822f531469cf860dfa5d010c87b284b747 (PR#180)

AlexanderAmelkin, 2020-08-07:
@yontalcar, indeed you did it. I missed that. It's understandable then why I can build with gcc-10 without problems. My changes on the branch just clean it up a bit further. 

Anyway, I don't get it then, what is the problem that @JoergFF is having there. Outdated sources?

BTW, Vaclav, I've sent you an e-mail about a week ago regarding CentOS ipmitool package, have you received it?

closed 2020-10-06 by AlexanderAmelkin