ipmitool/ipmitool #231: Customized ipmievd string output?

adamwoolhether, 2020-09-03:
Hi,

As the title suggests, is there a way to customize the output logged into rsyslog via ipmievd? The default information doesn't meet my requirements.

Ideally, the record ID # and timestamp (from bmc/SEL) can be included. I realize that freeipmi's ipmiseld allows this, but there is some incompatibility issues with our systems, preventing proper logging.

AlexanderAmelkin, 2020-09-04:
Unfortunatly, it is not currently customizable. You may add this functionality if you want. Take a look at `openipmi_read()` and `selwatch_read()` in `src/ipmievd.c`

AlexanderAmelkin, 2020-09-04:
Feel free to submit a PR

hthiery, 2020-09-26:
Hi @adamwoolhether, I had a similar problem that I solved by adding support to ipmievd to write the raw event into a socket. With that I could read all event on the other side.

Here is a link to that: https://github.com/hthiery/ipmitool/commit/68ad40907011e6ebd31215e7e513f20810f30c94

AlexanderAmelkin, 2020-09-26:
@hthiery, would you make a Pull Request from it?

hthiery, 2020-10-01:
Just opened a PR #245  for that.